hneemann/Digital #521: Strange Endianness Switching in ROM

ProxyPlayerHD, 2020-09-26:
this is a confusing issue.

for some reason loading a raw binary file into a ROM switches the Endianness of the contents. despite a RAW binary file not having a specified Endianness... it's just raw data.
for example 0x0002 in the Binary file gets turned into 0x0200 in the ROM.

Honestly i really dislike that, it should always be assumed that the Binary file is already in the correct Endianness (if there is one at all) and therefore should always be loaded straight into the ROM without switching around any bytes.

overall loading/storing from ROMs is really strange.
it supports loading multiple different formats (Raw Binary, modified Logisim "v2.0 raw", and i assume also Intel HEX) but always saves in the same format (Modified Logisim "v2.0 raw"), no matter what you set the file extension to.

hneemann, 2020-09-26:
What are the first two bytes in the binary file? And how do they end up in the first 16-bit word of the ROM?

ProxyPlayerHD, 2020-09-26:
the first 2 bytes are 0x00 and 0xA2 and in the ROM they are 0xA2 and 0x00.
same with any other pair of bytes

hneemann, 2020-09-26:
But this means that 0x00 is the first byte in the file and ends as the low byte of the first word in the ROM. 
This behaviour sounds reasonable to me.

ProxyPlayerHD, 2020-09-26:
this is what i mean with assuming Endianness.

you assume that the first byte (0x00) is the low byte and the second one (0xA2) is the high byte.
despite the fact that when you combine them into a 16 bit word in order that they appear in the file (0x00A2) it shows that 0x00 is the high byte and 0xA2 is the low byte.

so the code that loads the ROM assumes the binary file is Litle Endian while the file itself is always stored as Big Endian.

ironically this is only a problem with raw binary files, the Logisim "v2.0 raw" format is also Big Endian and loads normally into a ROM.
```
v2.0 raw
00A2
```
gets loaded as 0x00A2 into the ROM, as expected,

hneemann, 2020-09-26:
> ironically this is only a problem with raw binary files, the Logisim "v2.0 raw" format is also Big Endian and loads normally into a ROM.

It only seems so! The Logisim hex file stores words, not bytes. This is a completely different story! 
A Logisim file with the following content 

```
v2.0 raw
1
2

```
would be read as:

```
0x0001
0x0002
```
In a 32-bit ROM it would be read as:
```
0x00000001
0x00000002
```

A binary file, on the other hand, stores single bytes that cannot simply be considered as single words. These bytes must be assembled into words. 
And here it makes sense to consider them as little endian, because it seems reasonable to use the first byte from the file as the lowest byte in the first word. In the end the first word is stored at the lowest adress too.

But one could think about adding a big-endian option.

ProxyPlayerHD, 2020-09-26:
but wouldn't it make more sense to use Big Endianness, so that the order of bytes in the file is preserved when mushing it together into a word?
so that `0xDE 0xAD 0xBE 0xEF` stays the same order when turned into `0xDEADBEEF`.
that is exactly how it works in Logisim for example, and it just works perfectly fine.

personally i've rarely seen programs handle binary files in Little Endian.
only ones that i can think of that use Little Endian for words are 6502 Assemblers.

so yea maybe an option in the Settings to use either Big or Little Endian for Binary Files would be useful.

Overall this was a confusing journey, i don't know how i never came across this quirk before.

hneemann, 2020-09-26:
> but wouldn't it make more sense to use Big Endianness [...]

This is more of a philosophical question: what is more reasonable? Big-endian or little-endian. 

But to be honest, I think that most files that are imported are created by the users themselves with their own script. And I suppose that most users write little endian files. And if they write big endian, they will simply change their code when they realize that Digital interprets the import as little endian. It will only cost them seconds. So it doesn't seem worth the effort to provide another option.

Where do your files come from?

ProxyPlayerHD, 2020-09-26:
my files are either generate by [CustomASM](https://github.com/hlorenzi/customasm/), exported from Logisim, or made by hand in a Hex Editor.

all 3 use Big Endian, and I assume that that would be the most common way to generate Raw Binary files since it's the way Humans read Numbers (you don't write "1000" as "0010" so humans are naturally Big Endian biased)

ProxyPlayerHD, 2020-09-26:
also sorry for being a bit off-topic but i don't know of this is enough for it's own issue:
what exactly does this mean?
[image]
i can zip up and upload the whole project if you need it to replicate this.

hneemann, 2020-09-26:
This seems to be a real issue, which points to a buggy Verilog template.
But it's hard to identify the buggy template without the project. 
If you don't like to upload your project you can send me an [e-mail](mailto:digital-simulator@web.de).

hneemann, 2020-09-26:
Thinking about the problem I had an idea:
If your project contains multiple ROM components please ensure that the labels are unique!

ProxyPlayerHD, 2020-09-26:
here you go: [RISC-CPU-DIGITAL.zip](https://github.com/hneemann/Digital/files/5287027/RISC-CPU-DIGITAL.zip)

hneemann, 2020-09-26:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

ProxyPlayerHD, 2020-09-26:
alright it worked!
and now i got a small test program running on my FPGA Dev Board.
~1000 LE's is a bit large for a RISC CPU but it works and that's all that matters right now.
Thanks for the help

closed 2020-10-01 by hneemann