hneemann/Digital #524: Language to made case test

abnerbarros, 2020-10-02:
What is the language that you use to made test case to the components? 
Where may I found a tutorial to create more elaborated  tests?

hneemann, 2020-10-02:
The test case editor has a help button.
Pressing this button opens a help text which explains all the features of the test case parser.

abnerbarros, 2020-10-05:
In the help we have only a few examples that with a repeat command, a for command and a loop command. It also has the let command and nothing else. Do you have a formal specification of this scripting language? Has it been defined specifically for your program or is it derived from another language?

hneemann, 2020-10-05:
The parser is not based on a existing scripting language. It was defined specifically for this application.
And the parser is not created by a tool like javacc or antlr using a formal EBNF specification.
Because the parser is so simple it was created by hand. 
There are no more instructions than described in the help.

closed 2020-10-11 by hneemann