hneemann/Digital #526: 74xx library not available on mac

mbmartin44, 2020-10-06:
Hello. I am having an issue with the 74xx components library on mac. 

(On mac, I am running the application by directly using the .jar file, and have not found a different way to run it.)

When I go to hover over the "lib" selection in the components drop-down menu, nothing happens. I have tried clicking it, going to branch-view, deleting the software and redownloading, and a few other things, but nothing works. 

I really need to be able to use these 74xx components. Does anyone have any idea what my issue might be? 

Thank you for any help.

hneemann, 2020-10-06:
I don't know why the automatic detection of the lib folder doesn't work on your Mac, but it is possible to select the lib folder manually.

Click "Edit->Settings->Advanced" 
The first entry "Library" should point to the lib folder. 
If it does not, select the lib folder manually.

Before you do this, please copy the folder set there and add it together with the correct folder to this ticket.
Then I can try to find out why it did not work automatically.

mbmartin44, 2020-10-08:
i apologize for my own stupidity, but the reason that it wasn't working was that I had the Digital folder on my desktop. Moving the folder into applications alleviated the problem, though I'm not sure why it wasn't working since the pointer was correct for its original location. 
Thank you, your suggestion led me to the solution so I appreciate it.

closed 2020-10-08 by mbmartin44

JAicewizard, 2021-05-13:
From this issue I am make up that there is suposed to be a "Library" section in the components list, but it doesnt exist for me.

I have "Library" pointed to `/usr/share/java/digital-git`, which contains one file named `Digital.jar` (about 3.5 MiB in size). I do have read permissions, so thats not the problem.

I can try to upload the jar file to google drive so you can check it out?

JAicewizard, 2021-05-13:
Sorry, this seems to be a distribution problem. My distribution doesnt pack the library for the git version of Digital. I will report that to them.

hneemann, 2021-05-13:
@JAicewizard 
Which distribution do you use?

JAicewizard, 2021-05-13:
Arch linux.

Here is the link for the package: https://aur.archlinux.org/packages/digital-git/#news
And the one for the install script: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=digital-git

It is missing quite a bit, it seem that MIME types are also missing etc.

hneemann, 2021-05-13:
@JAicewizard 
Now I remember a corresponding discussion: #289
Possibly the package has its origin there.
However, I did not follow this further.