hneemann/Digital #535: test case bits(64,1<<n)

d-teif, 2020-10-16:
This test if working 
                    e         s31 s62 .... s0
repeat(32) (n) bits(64,1<<n)

But if we want to create 64 bits test, the value generate bits (64,1<<n) is incorrect (lower bits ar always to 1)

                    e         s63 s62 .... s0
repeat(64) (n) bits(64,1<<n)

Thanks for this great work 

Didier

hneemann, 2020-10-16:
What a crazy test case! But thanks for reporting it.

closed 2020-10-16 by hneemann

d-teif, 2020-10-16:
It's just to test a 64 decoder with 64 output who are set one at a time. I use it in mips architecture

hneemann, 2020-10-16:
You are right: Maybe it is not so crazy to build a circuit with 64 outputs. It is just that I have not done that yet. The header line in the test case gets really long. :smile:

commit 661bd7f4 (2020-10-16): fixes a bug in the test case bits(n,m) command if n is 64; fixes #535