hneemann/Digital #536: Higher dimension Karnaugh maps (5 and 6 variables)

maltaisn, 2020-10-19:
The Karnaugh table concept can be extended with 5 and 6 variables, becoming some sort of Karnaugh cube. A cube isn't very practical for visualization though, so instead each layer can be shown in a separate 2D Karnaugh map. Extra variables headers are shown on the side.

I took a few hours to implement the concept with 5 variables, it looks like this:
[image]

I can extend it to 6 variables and send a pull request if you're interested.

hneemann, 2020-10-22:
I am unhappy with the quality of the code of the current K-Map implementation. So I don't want to complicate the code any further before I have cleaned it up. Unfortunately I don't really have the time to do this at the moment.
However, the idea is good and I also think that this would be a nice extension.
But an extension to more than 5 or maybe 6 variables doesn't make much sense to me. No student will ever create such a K-Map by hand. If you want to minimize equations with so many variables by hand, you will rather use QMC.

callegar, 2023-11-01:
Would also be nice to have an option to collect the zeros rather than the ones, in order to visualize minimal product-of-sums expressions too.

callegar, 2023-11-01:
See also [1127](https://github.com/hneemann/Digital/issues/1227)