hneemann/Digital #537: Input Ordering Layout Bug?

wmuece, 2020-10-23:
It seems that the Circuit specific settings "default" layout of the decoder example below will not follow the vertical order of the inputs unless the settings is changed to "layout", in which case the vertical order is then followed.
[image]
[image]

[hneemann.zip](https://github.com/hneemann/Digital/files/5429436/hneemann.zip)

hneemann, 2020-10-23:
In the default shape, the order of the pins is defined via the menu item "Edit->Order inputs".
This allows the use of a pin ordering sequence which is independent of the circuit layout.

closed 2020-10-28 by hneemann

wmuece, 2020-11-09:
Helmut:

I was exploring how to export VHDL from your ProcessorIntr.dig

(P.S.  Your examples are excellent, and amazing)

I first removed the RAM and made the D line an external output.

When I attempted an export,  Digital gave me three "multiple output connected" errors for the D line and the two MUXA&B outputs.

I removed the D line error by simply eliminating the D tunnel. (Maybe that was a fake remedy because there are tri-state devices driving D)

For the MUXA&B output errors I really don’t see what the problem is.

Any comments/suggestions?

- Dean Johnson

hneemann, 2020-11-09:
The example processor uses a tri-state bus for data exchange between RAM, IO and the registers. This would require bidirectional ports in the HDL export. But this is not supported by the HDL code generators. 
Therefore there is a special processor variant (ProcessorMux.dig) that does not have a tri-state bus and uses a muxer instead.

PS: As far as I know tri-state busses are also not supported by FPGAs and, if used, are converted to a muxer structure by the synthesis tool. See also #184

wmuece, 2020-11-10:
Helmut:  OK, very good.
Can you share the instruction set or any other information that you might have on ProcessorIntr/ProcessorMUX?

Thanks. - Dean Johnson

hneemann, 2020-11-10:
@wmuece 
The [assembler](https://github.com/hneemann/Assembler) release contains a pdf with a instruction set description.