hneemann/Digital #538: Example of use with icarus verilog

abnerbarros, 2020-10-23:
I am looking for a example off use a component developed in verilog language. The help that have in external component is no suficient to teach how we be able to create a new project that use verilog component.

hneemann, 2020-10-23:
Add a "External" component found in the misc. menu and select iVerilog in the combo box.
But Icarus Verilog is not included with Digital. So you must install it yourself and select the iverilog binary in Digital's preferences to run Icarus Verilog from Digital.

abnerbarros, 2020-10-23:
Ok, I have did it. My problem is how to configure the external component and how and where imput my verilog code.

abnerbarros, 2020-10-23:
I have tried a very simple component but do not work.
[image]

hneemann, 2020-10-23:
Here is a simple working example: [z.dig.zip](https://github.com/hneemann/Digital/files/5430414/z.dig.zip)

abnerbarros, 2020-10-23:
Great! Thanks a lot.

hneemann, 2020-10-23:
You are welcome! :smile:

closed 2020-10-23 by hneemann

Johnlon, 2022-09-27:
Could we have documentation on how to use the component please