hneemann/Digital #539: Text size jumps around when zooming

maltaisn, 2020-10-25:
The text size jumps around depending on the zoom level. This can be annoying because sometimes the text doesn't overlap the circuit for certain zoom levels, and for others it does. Would it be possible to add this line in [`GraphicSwing.java`](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/draw/graphics/GraphicSwing.java#L45) to make font size consistent across zoom levels?
```java
gr.setRenderingHint(RenderingHints.KEY_FRACTIONALMETRICS, RenderingHints.VALUE_FRACTIONALMETRICS_ON);
```
I don't think this will affect performance too much. It also seems to slightly improve text rendering at small sizes.

hneemann, 2020-10-25:
Which OS do you use? On my machine it makes no difference at all.

maltaisn, 2020-10-25:
Ubuntu with OpenJDK 11. The same happens on Windows (still with OpenJDK 11). Here's what happens so you can compare:
- Current:
[image]
- With fractional metrics:
[image]

hneemann, 2020-10-25:
Would you please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) build?

maltaisn, 2020-10-25:
It works correctly in the latest build.

closed 2020-10-25 by hneemann