hneemann/Digital #540: 74190 Library Addition

HighEagle777, 2020-10-27:
version: v0.24
74190: Synchronous 4-Bit Decade Counter with Down/Up Control
[74190.zip](https://github.com/hneemann/Digital/files/5444955/74190.zip)

hneemann, 2020-10-27:
Thanks for your contribution. 
But it looks like the count test fails.

HighEagle777, 2020-10-27:
Hi Helmut

 

Thank you for an AWESOME product!

 

My apologies… I incorrectly copied the count test from74191 (a binary counter)

74190 is a decade counter.

Herewith the corrected test. I trust it works now.

 

Kind regards

Allan

HighEagle777, 2020-10-27:
Thank you for an AWESOME product!

My apologies… I incorrectly copied the count test from74191 (a binary counter)
74190 is a decade counter.
Herewith the corrected test. I trust it works now.
[74190.zip](https://github.com/hneemann/Digital/files/5445741/74190.zip)