hneemann/Digital #541: EEPROM Circuit Programming save option

HighEagle777, 2020-10-28:
version: v0.25
Again, thank you for an excellent software and the attention given to ensure it's success.

When an EEPROM is edited through the interface; ( [Right click Mouse] on the EEPROM, [Edit]. change some of the contents and click [OK], [OK] ), then the [Save] icon (Ctrl-S) is activated in order to save the changes.

However, if the contents of EEPROM is changed programmatically by writing via a circuit during Simulation, the [Save] icon is not activated. This implies that there have been no changes that need saving even though the contents of the EEPROM have been changed. If one closes Digital at this point, the changes made during the Simulation are lost.

It seems the only way to save the new contents after having been written during Simulation is to click [File], [Save As] and select the .dig you are working with to overwrite.

Is it possible to activate the [Save] option when the contents of the EEPROM are changed during Simulation?

hneemann, 2020-11-01:
That's a tricky one! 
I have to think again about how this can best be solved.

closed 2020-11-12 by hneemann

hneemann, 2020-11-12:
@HighEagle777 
Please feel free to try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) version.

HighEagle777, 2020-11-13:
Appears to be working well. Thank you!

commit d8dadcc2 (2020-11-12): better handling of model driven eeprom modifications, closes #541