hneemann/Digital #542: Compatibility with Java 8 u271

jwhsu01, 2020-10-29:
Updated to Java 8 rev 271 today. Digital still works, however, I now cannot access the "Library" folder under the "Components" menu. I can still access all the other folders and the "Custom" folder. Was something changed in the latest version of Java which prevented access to the "Library" folder?

hneemann, 2020-10-29:
Not that I'm aware of!
But i have moved the location of the library folder in v0.25. 
If this causes the problems you can go to Edit->Settings->Advanced and select your library location in the top most combo box.

jwhsu01, 2020-10-29:
Got it to work by moving the "lib" folder to a different location. It is weird because on some directories, the dialogue box is completely blank, while in others, included folders show up for selection. (BTW, this is on OSX Mojave)

hneemann, 2020-11-01:
This problem seems to be solved. If there are more problems, just open it again!

closed 2020-11-01 by hneemann