hneemann/Digital #543: Create parameterized circuit for generic circuits

maltaisn, 2020-10-30:
I've been making a lot of generic circuits lately and I found them hard to debug. Since the generic parameterization is only applied when the component is used as a subcircuit, there's no error highlight when the model fails to be created, only a  somewhat vague error message. It's especially frustrating not being able to see what the parameterized circuit looks like.

So I added the following feature in my fork: Edit > Create parameterized circuit shows a dialog to enter generic code and then generates a new circuit parameterized accordingly. The circuit can then be simulated and errors are much easier to find, then fix on the original. I had originally made it so that the circuit was modified in place but that was cumbersome. The generic code for generic subcircuits is changed in the process to be just a list of `export` of all the variables contained in the context at the time of generic resolution.

Also I noticed that test cases have an apparently unused generic attribute like the rest of the component in a generic circuit, so I made use of it to allow testing a circuit with specific arguments.

Both features are working great for me. Do you think that would be a good idea?

The fork is [here](https://github.com/maltaisn/Digital/tree/apply-generics) in case you want to test, on the `apply-generics` branch.

hneemann, 2020-11-01:
That are both good ideas. I have to think about them!

hneemann, 2020-11-01:
What about adding some generic code somewhere to the circuit that is used for generic parameterization when the circuit is started directly and that is ignored when the circuit is used as an embedded circuit?
There could be a field in the circuit specific settings or there could be a special component to add the code similar to a test case.

maltaisn, 2020-11-01:
That would allow to get rid of the `isPresent(args)` blocks since `args` would always be present. A dialog could be shown on the first simulation or when the parameters change to set this field.

But ideally, the circuit itself should be modified after the simulation. Otherwise debugging would still be hard, since the resulting circuit would only be shown if there's no errors in it. For example, there's a bug in the generic code of the following splitter, it causes a mismatch in the number of input and output bits:
```
this.'Input Splitting' = (args.dataBits - 2) + ",1";
this.'Output Splitting' = "" + args.dataBits;
```
Running the circuit would immediately show the error: "Bit count of splitter is not matching. Affected are: Signal 0-X, Signal out" or something similar, which doesn't really tell anything about what the parameters actually were. In this case the bug is pretty obvious, but sometimes it's much harder, like when the splitting has several parts. Whereas if the circuit can been viewed separately, all the values can be seen.

If the circuit has generic subcircuits, there's no would be no direct way to check what its parameters actually were. And if like I said the circuit is modified after the simulation to export the values that were used, then the original code is lost, maybe even the original component if `setCircuit` was used. That's the same issue I encountered when I modified the circuit in place. There are ways around this, like putting the original code in comment, but that's still not great.

It also happened once that I needed to copy part of a generic circuit to a non-generic circuit. This was straightforward: create the circuit with the required parameters, and copy it. If the parameterized circuit is only available during the simulation, this can't be done.

The two ideas aren't exclusive though, both could be implemented.

hneemann, 2020-11-01:
Hmm! You are right!
Maybe one should add a component in which generic code for the direct start of the circuit can be entered, and this component could then have a button by which one can create a concrete circuit, which can be used if there are errors that are hard to track down.

maltaisn, 2020-11-02:
I'm not sure a component is the right solution though. What happens if the user puts two of those components?

And what happens if there's no such component in the circuit? The current behavior is to simulate as if the circuit wasn't generic, using the actual parameter values. But having to manage both the actual parameters and the generic code is extra overhead and a source of inconsistency. Sometimes one can work and not the other. I believe it would be best if a generic circuit was defined by the generic code at all times, so either it fully works, or it doesn't at all.

jay20162016, 2020-11-02:
It seems that somehow, I was able to drag the generic circuit as an embedded circuit (is this intended?), leading to easy debugging.

hneemann, 2020-11-03:
@maltaisn 
> I believe it would be best if a generic circuit was defined by the generic code at all times, so either it fully works, or it doesn't at all.

I agree! 
And I think that a component with init code is a good solution to ensure this!
Either the parameters come from outside when the circuit is embedded or they are provided by the component when the circuit is started directly. But there are always parameters that define the state of the circuit.

If this component is missing, the direct start is not possible. If you insert several such components, only one of them may be activated at a time. This allows you to use more than one, between which you can easily switch.

The branch [genericInit](https://github.com/hneemann/Digital/tree/genericInit) contains a first implementation to play with.

maltaisn, 2020-11-03:
Being able to switch between different parameters is certainly useful. I just tried it and it works great!

hneemann, 2020-11-04:
@maltaisn
Now you can also create a concrete circuit. It would be very helpful if you could test this new function a bit.

maltaisn, 2020-11-04:
When you create a concrete circuit, `ResolveGenerics` set the `genericArgs` field on generic elements, so if the circuit has generic subcircuits, it appears to work fine. But this field is not serialized, so by saving the file, closing it and then reopening it, the simulation fails. Serializing it would be one option, but instead I think it would be best to modify the generic code of the subcircuits. The original code has no purpose anyway and this would allow the user to see the concrete values used for the parameters.

Also the indentation in `barrelShifter.dig` is messed up.

hneemann, 2020-11-05:
@maltaisn 
Creating a editable, serializable, parameterized circuit was much more complicated than I had initially thought. 
Especially if recursive circuits have to be considered. :-(

hneemann, 2020-11-06:
closed in 0184e2c43cd288befaddcad8f8b966a6c557719f

closed 2020-11-06 by hneemann

maltaisn, 2020-11-06:
Thank you for implementing it!

For tests, do you intend on allowing the generic code for the circuit to be set for each test like I proposed (with fallback to generic component if none set)? Also there's a minor issue with strings in created concrete circuits. Their value isn't escaped, so characters like `"`, `\`, etc. can't be used.

I'll send a pull request to update the French translation.

hneemann, 2020-11-07:
@maltaisn 
> For tests, do you intend on allowing the generic code for the circuit to be set for each test like I proposed (with fallback to generic component if none set)?

This function is quite difficult to implement. So far, when a circuit is translated into a model, there is only one possible model. Even if a generic circuit is translated into a model, there is only one possible model defined by the one enabled generic init code component.
If the test cases in a circuit each required their own specific model, this would no longer be the case. 
And this makes especially the export of test cases to VHDL/Verilog much more complicated, because not only one single VHDL/Verilog file has to be created, but each test needs its own specific VHDL/Verilog file.
In addition, the benefit is rather small, since it is already possible to create different tests for a generic circuit with different parameterizations, as the barrel shifter example shows.

hneemann, 2020-11-07:
@maltaisn 
Today a had the time to play around with test cases, which can parameterize the circuit specifically for their test.
This feels so useful and natural that I will add this feature after all.
But I am not sure if I will implement the export of these test cases to VHDL/Verilog. 
The branch genericTestCases contains the implementation.

maltaisn, 2020-11-07:
I don't know a lot about HDL, I hadn't thought about it when I implemented it in my fork. 
I suppose it's probably not a big deal to ignore the generic tests during export for now. 

Everything works for me in the new branch. I've updated the French translation.

hneemann, 2020-11-08:
>  I've updated the French translation.

It is much easier for me if you update the translation by completing this file and then sending it back to me.

[lang_diff_fr.xml.zip](https://github.com/hneemann/Digital/files/5506220/lang_diff_fr.xml.zip)

maltaisn, 2020-11-08:
Sure, here's the file: [lang_diff_fr.zip](https://github.com/hneemann/Digital/files/5506257/lang_diff_fr.zip).
I followed what was said in the said in the [how to](https://github.com/hneemann/Digital/blob/master/src/main/resources/lang/howTo.md) and imported it with `LanguageUpdater`. 
I figured the rest would be relatively straightforward matter of fetching & merging.

hneemann, 2020-11-08:
You are right. I just didn't realize that you did it the intended way. :-)

maltaisn, 2020-11-11:
There's a bug with Set Input Defaults on generic circuits. It checks the actual values of the circuit instead of the shallow copy used by the simulation, so the defaults never change.

hneemann, 2021-01-07:
@maltaisn 
You have worked quite a lot with generic circuits. 
Please have a look at the branch [genModCirc](https://github.com/hneemann/Digital/tree/genModCirc), and there the examples in src/main/dig/generic/modify.
Would you say that this is a useful extension, or is it rather superfluous and too complicated?

maltaisn, 2021-01-07:
It could certainly be useful. It's a more versatile approach than recursive generic circuits. If anything, I think this simplifies the creation of generic circuits.

One issue I see here is that positioning is trial and error. Maybe there could be a position indicator in the status bar for generic circuits? Another issue is finding the internal name of components when using `addComponent`. The name of some components is not always obvious. It could probably be indicated in the Help dialog of the components like the names of  attributes are indicated.

hneemann, 2021-01-07:
I have added the component names to the help dialog and added a circle to the grid that indicates the origin of the coordinate system. This way you are able to place the components easier.
Meanwhile, I am also sure that this is a useful extension. I will merge it.