hneemann/Digital #545: Generic Circuits Parameterization doesn't work

jay20162016, 2020-11-03:
[image]
[image]
The semicolon is there, but the parameterization doesn't read it.

hneemann, 2020-11-03:
The error message says that there is an error in a component "Out (Y)". 
However, the attribute window shows the attributes of a component "NOT_1.dig". 
What does the generic code of the component in the specified component "Out (Y)" look like?
There should be a component "Out (Y)" in the file "NOT_1.dig" which contains the bug.

jay20162016, 2020-11-03:
oh, I seem to have forgotten a semicolon in NOT_1.dig. Thanks!

closed 2020-11-03 by jay20162016