hneemann/Digital #547: When running under java 14 the dropdown menu disappears immediately without interaction

VitalyAnkh, 2020-11-05:
As the title said. It's like the error long time ago: https://bugs.openjdk.java.net/browse/JDK-6427176

When I set the java runtime jdk 8, the problem doesn't exist. So when I use Digital I had to set the java runtime first for I usually use the latest jdk. How to solve this problem?

hneemann, 2020-11-06:
Can you please explain in more detail what you did and what happened?
I have tried jdk14 and 15, and everything seems to work as expected.

VitalyAnkh, 2020-11-07:
@hneemann After clicking the drop-down menus, they appear and disappear immediately so I cannot interact with them.
This is a screenshot video: https://youtu.be/G55W4HEshbk

hneemann, 2020-11-07:
The menus in Digital are the unmodified Swing components. 
If something doesn't work with them, it's not the Digital code, and I don't think it can be fixed there.
To be sure you can try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) version built on my machine.

hneemann, 2020-11-27:
@VitalyAnkh 
Was it possible to solve the problem, or is it still an open issue?

closed 2020-12-15 by hneemann

VitalyAnkh, 2020-12-17:
@hneemann Sorry for the late reply, I haven't check emails for a loog time.
It's still an open issue for me, but I could set Java runtime as Java 8 to avoid this problem. :smile: 
Digital is a great work! Thank you!