hneemann/Digital #550: Digital Can't See Folders

SteveMillman, 2020-11-08:
I'm on a Mac running Catalina 10.15.6. If I try to open or save to a folder, Digital can see the folders in my home directory and I can select and dive into them. However, if I want to save or load a file in a folder on my Desktop, I can go down to the Desktop but Digital can't see any folders there.

On my old Mac, running High Sierra and the same version of Digital, this is not a problem - I can see such folders.

Thank you!

Steve

version: v0.25

hneemann, 2020-11-08:
There is no special code in Digital that deals with a desktop folder, and I don't know if the desktop on a Mac is a real folder in the file system or not. If this is not the case, the files may not be listed. This might also depend on the JVM implementation.

This problem seems to be similar to #526

I have no idea how the Mac implements the desktop and how this has changed in the different OS versions, since I am not a Mac user myself. 
So my best advice is: Don't store any files on the desktop.
If someone knows better, please leave a comment.

SteveMillman, 2020-11-08:
I did some more checking and it's not just a problem for folders on the desktop. Digital also can't see folders in the Documents folder. But it can see folders in the Library folder and its subfolders. This is very strange. I'll poke around on Java sites to see if I can find anything. This may have something to do with the crazy permissions that Catalina introduced.

SteveMillman, 2020-11-08:
Here is a bunch more information... I called Apple support. Useless. They don't like helping with 3rd party apps.

The hint given above that this was similar to #526 seemed promising. I tried moving Digital to the Applications folder. No joy. Then I noticed that I also couldn't see the 74XX library. I poked around and the permissions on the subdirectories in the Digital folder were 700. So, if owned by the admin, I couldn't see them. I changed the permissions to 755 on all the folders all the way down and that problem was resolved. But, still no joy.

Then I decided to do something odd. I opened Digital and selected File->Open and screenshot #1 is what I see.

[image]

So I click on Desktop, which is where the folder holding my design resides. And I see this second screenshot:

[image]

Nothing there, right? So, on a whim, I typed Lab0, the name of the folder where my files are...

[image]

And then hit open...

[image]

Interesting, so it found it although it was invisible. So now I type the name of the file:

[image]

and click on open...

[image]

and success! Why is this happening? So the files and folders are invisible??? Interesting trick... There's got to be something with permissions going on somewhere.

SteveMillman, 2020-11-11:
Ok, now it's getting silly. I added another folder on my Desktop. Now Digital can see both folders. I give up - this seems like a bizarre workaround, but there it is. There's obviously a bug somewhere in Java or the Apple OS, but there you go. Thanks to all who thought about this issue.

SteveMillman, 2020-11-12:
Nope, I spoke too soon. While that workaround seems to be ok for finding files, there is a more serious issue. It appears that Digital finds submodules via the Custom choice under Components. And, from what I can tell, all the .dig files in the current folder are visible in the Custom submenu. But, It appears that Digital is unable to see the .dig files in the folders on Catalina. Again, it all seems to work just fine using the same tools on High Sierra. This may mean we'll be unable to use Digital if I can't find a work around. If anybody can help I'd really appreciate it!

SteveMillman, 2020-11-12:
Ok, more information. If I've save a circuit, first.dig, in a folder, and it's the only thing in the folder. If I then do File->New, there is no Custom choice under Components. But, now I have an empty design window. If I do File->Save As and save the empty file as second.dig, suddenly the Custom choice is back and both first and second are available. Why can Digital sometimes see things but sometimes not? I feel like I'm in the Twilight Zone...

hneemann, 2020-11-12:
I took a second look at the code: The dialog to open a file is the unmodified JFileChooser which comes with the JRE installation. Here everything should be visible which is also visible for the JRE.
But the "Custom" submenu in the "Components" menu scans all files in the directory of the opened file and all sub-folders. If during the scan the number of files reaches 5000, the scan is interrupted. In this case files might be missing.

SteveMillman, 2020-11-12:
Interesting. But, in my case, there were a few files in the directory. In fact, if the only other file in the directory was the one .dig file, I still couldn't see it unless I did the new tricks. Like starting in the correct directory or saving a new file immediately and then the other files would show up. I'm wondering if this is a Java vs MacOS issue... I wouldn't know how to pursue that. Overall, I really like what you've done with Digital. Thanks for you help and please let me know of any suggestions you may have to get this resolved.

SteveMillman, 2020-11-15:
I noticed the same issue of being unable to see .dig files on a Windows machine in the Components->custom menu. The same workaround does the trick: open a new file and do File->Save As in the directory in question. And, like magic, previously created .dig files are suddenly visible in the custom menu.

hneemann, 2020-11-15:
Hm! The custom menu has a last entry called "Update".
Does pressing this button make a difference?

I also thought about upper screenshots. What happens if "All Files" is 
selected in the file selectors combo box instead of "Circuit"?

SteveMillman, 2020-11-15:
If the custom menu is there, selecting Update makes no difference. But, if I recall, the custom menu was only there and empty after a bunch of gyrations trying to get things working. And, when it appeared, Update was there but clicking it did nothing. However, as I said, the custom menu is normally not present. However, as soon as I Save As the new, empty design, the custom menu appears as do all the .dig files.

Selecting All Files makes no difference when trying to open a file: it still shows nothing to select.

I neglected to say in my prior post that I was on a Windows 10 machine.

hneemann, 2020-11-15:
> However, as I said, the custom menu is normally not present. However, as soon as I Save As the new, empty design, the custom menu appears as do all the .dig files.

This is the intended behavior. The cusom menu shows all files that are in the same folder as the current file. If the current file has not yet been saved, there is no folder yet and the menu does not exist.
Pressing the "Update" button should not actually change anything, as it simply rescans the folder. 
It is normally only needed when files are moved, copied or renamed externally with a file explorer.

SteveMillman, 2020-11-15:
Ah, that would explain the custom menu question. Glad that is resolved! So back to just the mystery of not being able to navigate to the folder without blindly typing in folder names... Thank you!

hneemann, 2020-11-15:
But that brings up another question: If the jvm can see the files when the custom folder is scanned and they appear in the custom menu, why should the jvm not be able to see the files when using the JFileCooser?
The whole thing seems more and more strange to me.

flandreas, 2020-11-15:
Have you checked the security settings in your MacOS settings? Sorry for the screenshot being in german.

[image]

If I disable "Desktop" for the application "java", I can almost reproduce your symptoms. Folders on the Desktop are not shown, but I can navigate into the folder by typing the directory name. However, I am not able to open a circuit, instead I get an Error dialog "Operation not permitted".

If I create a new folder on the desktop using the "house icon" in the file dialog, I am able to store circuits in this new folder, as well as to open the circuit again.

SteveMillman, 2020-11-15:
I knew about the permissions thing and have it set. I can see folders on the Desktop and in Library, etc. But not in Documents or on the Desktop. I did not have jar selected here so I added it and it made no difference. So I took it back off and just left Java.

Unfortunately, while doing this work, I hit another strange snag - if I have a hierarchical design and open the top level, Digital complains that the submodules are missing and the custom menu is empty. Grrr! If I then make a tiny change to the design and save it, suddenly all the submodules are present and the custom menu is populated.

It appears that when I opened the top-level design with the command:

java -jar /Applications/Digital/Digital.jar four_bit_adder.dig

Digital doesn't realize what folder I'm in and, therefore, the custom menu can't find it. So, I make the tiny change and save it, but it saves in my home directory... So it has to be a save as by navigating down to the directory where I'm working.

Is there a way to make it so Digital knows where I started java? It works ok if I specify the entire path when I open the design:

java -jar /Applications/Digital/Digital.jar ~/Desktop/Lab1/four_bit_adder.dig&

Maybe that's the answer on Mac for now... Always start with the full command and full paths rather than double clicking on Digital.jar... When I open this way, and try Open and go up to my home directory, I can navigate back down as one would expect - all subfolders are properly visible.

d-teif, 2020-12-17:
I had the same problem with my mac bookair. (catalina 10.15.7)
When I start directly Digital by clicking on Digital.jar the ~/Desktop directory look empty from digital circuits insteed 2 dig file are presents! 

When I run it from terminal from Desktop java -jar Digital.jar it work perfectly, i see these 2 files.

I noticed it's only for file in ~/Desktop, ~/Download, but for others like ~/Documents or something else it work normaly. 

Look like an os problem.

DT

helderdaniel, 2021-01-23:
I personally haven't experienced this problem with Catalina, however some friends had and some don't.
We still did not figure out why it is happening only to some of us.
The workaround that seems to work for all of us with the problem, is starting digital from terminal, as @d-teif reported:

 java -jar Digital.jar 

Some of us use Digital also in Windows and Linux.
This issue only came up with mac Catalina.
We did not check with previous versions of OS X.

sniptamallick, 2021-02-23:
Ran into a similar issue mentioned in this thread. When I tried to run Digital on my Mac by clicking on the jar file included in the zip file, I could not see any of the subdirectories in the Library like the 74xx library. After running Digital from terminal using "java -jar Digital.jar", I can see everything now.

peterb12, 2021-03-29:
Similar things happen with Logisim.  I think this is an interaction with the JRE and the security model. We should probably look beyond the four corners of Digital to see if there's a best practice for dealing with this beyond the "start from the command line" workaround.

peterb12, 2021-03-29:
I was able to trivially create a launchable `Digital.app` on my Mac by running the following command line from the installed copy of the app:

`jpackage --name Digital --input . --main-jar Digital.jar --jlink-options --bind-services`

Sadly, I don't know enough about Maven at the moment to propose how to integrate this into the app's formal build process, but I think we should try it out if possible. 

I followed the instructions from here: https://docs.oracle.com/en/java/javase/16/jpackage/packaging-overview.html#GUID-C0AAEB7D-1FAB-4E20-B52C-E2401AC2BABE

hneemann, 2021-03-29:
This is not as simple as it seems because the documentation says: Quote:
"To package your application for multiple platforms, you must run the packaging tool on each platform."

peterb12, 2021-03-29:
Yes - I don't think this can or should replace distributing the jar file for Linux; it's just providing a clickable app for MacOS that addresses the security model problems that currently exist on that platform.

hneemann, 2021-03-29:
The problem is much more that the builds for the distributions are generated by travis-ci, and I'm not sure if multiple virtual machines can be launched for a single distribution. But at least travis-ci offers MacOS VMs.

helderdaniel, 2021-11-10:
Below is a way make sure *.dig files are shown in mac and also to associate *.dig files with Digital, without needing to open Digital from command line.

A script, based on _Digital.sh_, is set so that when double-clicked in Finder opens Digital. (Digital.sh needs to be modified, since _realpath_ is not available in Mac OS X.)
Then an Automator is created to open *.dig files with Digital, double-clicking them in Finder.

**Part 1: Create a script to run Digital when double-clicked in Finder**

**#1**
Download latest Digital release from:

https://github.com/hneemann/Digital

**#2**
Unzip it and move folder to:

/Applications/Digital

**#3**
copy:

/Applications/Digital/Digital.sh

to:

/Applications/Digital/runmac.sh

**#4**
edit: /Applications/Digital/runmac.sh

paste below text and save:

```
#!/bin/bash
realpath() {
    [[ $1 = /* ]] && echo "$1" || echo "$PWD/${1#./}"
}
```

**#5**
In Finder, right-click "runmac.sh" and select "Open with" and then "Other..."

In Enable: change "Recommended Applications" to "All Applications"

Check "Always Open With"

Select to open with: /Applications/Utilities/Terminal

**#6**
Now to run Digital, on Finder double-click on: /Applications/Digital/runmac.sh

**Part 2: Associate** ***.dig files with Digital, to be opened when double-clicked in Finder**

**#7**
Run: /Applications/Automator

In "Choose type for your document:",choose: Application

Now select: "Run Shell Script"

In "Shell:", select: "/bin/bash"

In "Pass input:", select "as arguments"

In the text area below paste:

```
for f in "$@"
do
  /Applications/Digital/runmac.sh "$f"
done
```

Finally go to File->save and save as: "/Applications/Digital/openmac"

**#8**
Now it is needed to associate *.dig files with "openmac",so:

In Finder, right-click some *.dig file and select "Open with" and then "Other..."

Check "Always Open With"

Select to open with: /Applications/Digital/openmac

**#9**
Now double-clicking on a *.dig file will open it in Digital

_Notes:_
It should not be necessary, since copying from Digital.sh executable will generate an executable, but to make sure "runmac.sh" is executable, open a terminal, go to: "/Applications/Digital" folder and issue the command:

chmod +x runmac.sh

**Biblio**
Realpath implementation on Mac:

https://stackoverflow.com/questions/3572030/bash-script-absolute-path-with-os-x

Use Automator to open files with shell scripts on Mac:

https://superuser.com/questions/239231/osx-open-with-bash-script

GreyCon, 2023-02-28:
Hi, this looks like a permissions issue on the parent folder. I created a folder called DigitalWork in my home directory, then set its permissions wide open;

cd 
mkdir DigitalWork
sudo chmod 766 DigitalWork

I then save my various circuits, including embedded ones in this folder, and all works OK, including the custom list under Components.

hneemann, 2023-02-28:
@GreyCon 
Thanks for sharing.

GreyCon, 2023-02-28:
@hneemann A pleasure. Phenomenal piece of work Helmut. I've been using Logisim for a homebrew CPU project, and ran into some limitations in terms of propagation delay modelling in sub-circuits - looks like you have addressed that, so I'm trying to get to grips with Digital today. Some things seem a bit quirky, but that's my bad - trying to unlearn the Logisim mental model.   Many thanks for this.

Saij, 2023-05-01:
Regarding the building process: you could maybe move to Github Actions. I already have implemented a build process for my fork of PrusaSlicer which outputs binaries for all 3 systems - built in parallel.

mhsprang, 2024-01-12:
Late to the party, but I just suffered from the same problem. I found the solution on StackExchange:
https://stackoverflow.com/questions/65469536/why-does-a-jar-file-have-no-permissions-to-read-from-disk-when-started-via-doubl

On newer macOSses, Jar Launcher.app is called JavaLauncher. After giving this app Full Disk Access through the System Preferences, I could start Digital again by just double-clicking on its icon.