hneemann/Digital #555: Custom section of component tree view not updated when library is updated

maltaisn, 2020-11-13:
There seems to be a bug with the Custom section of the component tree view, it isn't updated when the library listener is called. In [`LibraryTreeModel`](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/gui/components/tree/LibraryTreeModel.java), the container map already contains an entry for the section so no new container is created, leaving the old one unchanged. You can reproduce this by first expanding the Custom node to create the initial map entry, then making a change in the library.

hneemann, 2020-11-14:
Please describe in more detail what you have done. 
I have tried it on my machine and it seems to work as expected.

maltaisn, 2020-11-14:
I save an embedded circuit from a different window, then force the library update from the other window. The menu is updated correctly, but the tree view isn't.
[image]

If the Custom node hadn't been expanded in the first place, this wouldn't have happened, because [this map](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/gui/components/tree/LibraryTreeModel.java#L25) wouldn't have contained the node yet. But being expanded, the container for the node had already been created, and is never recreated.

xotmatrix, 2020-11-15:
I've struggled with this myself, constantly restarting to update the Component Tree. Finding the Component Menu could be updated saved my sanity.

commit a235e858 (2020-11-15): closes #555
closed 2020-11-15 by hneemann