hneemann/Digital #558: Is a High-Z constant possible and desirable?

xotmatrix, 2020-11-15:
I'm working on a bus-oriented design and sometimes I want to place some bus lines in a High-Z state through a Splitter/Merger. I've seen some similar questions and suggested posted here. I don't understand some of the simulation difficulties in addressing their issues but I did find some workarounds that allow my embedded circuits to communicate High-Z bit states through their bussed inputs and outputs.

The first workaround I tried was using a disabled Driver. This works OK and is nice for outputting multiple bits but its meaning in the schematic is confusing.

The second workaround I tried was using an open Switch. This works as well and is a little more clear but it takes up a lot of space, especially for multiple bits.

Then I landed on using a floating Fuse. It is as compact as a Constant Value but can't address more than one bit. I'm pretty happy with this solution although using a floating Fuse this way seems more like a bug than feature.
[image]

I found what I really wanted was for the Constant Value component to be able to express a High-Z state.

hneemann, 2020-11-16:
Although it is very easy to implement, it somehow feels wrong:

[image]

But your workaround also feels wrong and should be fixed somehow.

hneemann, 2020-11-18:
@xotmatrix 
Please try the latest pre release [v0.25-114-g6acf491](https://infdigital.dhbw-mosbach.de/).
It allows to flag a wire as "Not Connected". Please let me know how it works.

xotmatrix, 2020-11-19:
Seems good although I've noticed things are behaving a little differently with multiplexers. 

In v0.25 the multiplexer is turning Z into 0.
[image] [image]

In 114-g6acf491 the multiplexer is turning Z into seemingly random values that change each iteration.
[image]

I can't really say that either is wrong, it's just an observation. I do not believe a real multiplexer is capable of passing a Z input to its output.

hneemann, 2020-11-19:
That's the main reason why I don't like to allow open inputs in the first place:In my lab exercises, students sometimes rely on open inputs behaving as if they were connected to ground because that's what happens in the simulator. They are then very surprised that this is not the case in a real circuit.  In a real circuit, an open input behaves in a way that is difficult for students to understand: Sometimes the input seems to be read as high, and a moment later it seems to be low. And sometimes this changes when you move your hand over the circuit.
So I thought that if I make open inputs more easily available, they have to behave more like a real open input: one cannot rely on a value.

hneemann, 2020-11-19:
closed in 57deba5465be35c9c22a9475a2ade297784014aa

closed 2020-11-19 by hneemann