hneemann/Digital #559: Two-connector LED has a broken resistor?

xotmatrix, 2020-11-15:
The two-connector LED looks like it has a resistor on its cathode but doesn't act quite like it when connected to ground. I was trying to use the LED as a bus status display as well as a pull-down resistor but it does not function that way. At first I thought its appearance was merely misleading. However, a pull-down resistor connected to the LED's cathode end does not function as expected. Seems like a bug but I also recognize the simulator's handling of passives is very limited.
[image]
[image]
[image]

hneemann, 2020-11-15:
Digital is not a simulator for analog circuits. 
If you want to simulate real resistors you have to use something like Spice.
In Digital, a pull-down resistor is just a flag to indicate that the network it is connected to has to be set to low when all other connections to the network are in the high-Z state. That's it.
The LED with series resistor is simply an indicator that has no other function. You can't use this LED e.g. to build a "wired or" or something like that either.

xotmatrix, 2020-11-15:
I understand. I can see its utility though its suggestive shape is confusing.

hneemann, 2020-11-16:
The shape was chosen that way because a real diode, if it is operated without a series resistor limiting the current, would be destroyed immediately.

closed 2020-11-16 by hneemann