hneemann/Digital #561: Freeze on Analysis on MacOS Big Sur 11.0.1

howtoosee, 2020-11-18:
[image]

Programme freezes when analysis tab is opened when user sets system preferences to always prefer tabs when opening new documents.

To reproduce:
1. System Preferences > General > Prefer tabs: always.
2. Open analysis for a circuit
3. Programme freezes

Programme version: Build git-version: v0.25

System info:
OS: MacOS Big Sur 11.0.1
Java: 
java 11.0.8 2020-07-14 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.8+10-LTS)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.8+10-LTS, mixed mode)

hneemann, 2020-11-18:
Does it freeze immediately after opening the dialog, or are there still actions possible/necessary until it freezes?

howtoosee, 2020-11-18:
Immediately after the dialogue opens.

hneemann, 2020-11-18:
At first look I can't find anything in the code that could be the cause.
Is there any processor load in the frozen state?

howtoosee, 2020-11-19:
There is a CPU spike from ~3% to as high as 60-80% when frozen

hneemann, 2020-11-19:
And after the spike the load of the Java process drops to zero?

howtoosee, 2020-11-19:
That doesn't seem to be the case. The process remains high consistently with some slight fluctuations. Java doesn't crash but simply freezes.