hneemann/Digital #562: Allow probe signals to be used in tests

maltaisn, 2020-11-19:
This might be a niche use case, but would it be possible to also allow using probes as test signals in the same manner we can use inputs and outputs? Probes can already have a label and can be used in the measurement graph so that would be a relatively small addition.

I already implemented it in my fork for my own use. I needed to have this for an ALU that outputs a number of condition flags as a single output, but not all of them have a meaningful value for all operations. I couldn't specify individual bits in tests and I didn't want to add outputs just for the purpose of testing. It could otherwise be useful to test any sort of intermediary values.

closed 2020-11-19 by hneemann

hneemann, 2020-11-20:
@maltaisn 
Although this was very easy to implement, it leads to another question: Wouldn't it be desirable if it were possible to check any condition in a test case? Shouldn't it be possible to solve your "ALU flag test problem" without probes?

maltaisn, 2020-11-20:
How would the targeted signal be identified? In my case the other solution I thought of would have been to allow testing individual bits in a output bus. This would be slightly simpler than having to put probes, but on on the other hand it would also be much less versatile. Probes allow testing any signal in a circuit, and a meaningful name can be associated to it.
What did you have in mind?

hneemann, 2020-11-20:
I am thinking of something like the declaration of a new signal, which can be used in a test case like this:

```
declare Z = (D>>3)&1;

A B Z
0 0 0
0 1 0
1 0 1
1 1 0
```
But this is not so easy to implement because a special parser is used for the header. Furthermore it feels redundant because signals are actually defined in the circuit and it would be strange if you could define signals in the test-case too. Maybe it is better to use a probe instead. Or you just create your own circuit for the tests, which embeds the circuit under test and generates all needed signals. This also does not cost any runtime in the simulation.

hneemann, 2020-11-21:
@maltaisn 
I have implemented a corresponding extension for testing purposes. (branch [virtualTestSignals](https://github.com/hneemann/Digital/compare/virtualTestSignals))
To leave the special header parser unchanged, the order of declaration is unusual: You first use the signal in the header and then define it afterwards.

```
A B Z

declare Z = (D>>3)&1;

0 0 0
0 1 0
1 0 1
1 1 0
```

It is not yet completely refined. So do not expect too much.

commit 3bee0738 (2020-11-19): allows to use probe values as test case outputs; closes #562