hneemann/Digital #564: Cannot generate HDL for pin-control with bit-width > 1

holstvoogd, 2020-11-21:
I've been working on getting some circuits running on an Arty-A7 board and think I've ran into a bug in the HDL generation.

When I try to generate the HDL, I get a pretty generic error: 'Error during export to VHDL.'
I have narrowed it down to having any output connected to a pin control through a splitter. The error and part of the circuit:
[image]

A few observations/notes:
- The pin-control is set to 8 bits width.
- Simulation works correctly.
- Having any output connected will trigger the error; no output connected = no error. 
- Single bit pin controls work fine.

I haven't found any log files or other output to help diagnose the issue further.

hneemann, 2020-11-21:
In the simulation, the PinControl component can be used without restrictions. When used in an HDL export, only one pin may be connected to the output. No component is allowed between PinControl and output. Also no splitter. 
But there should be a better error message.

holstvoogd, 2020-11-21:
Ah, makes sense. Thanks for the quick reply!

hneemann, 2020-11-21:
@holstvoogd 
I forgot to mention that for a multi bit output several pin numbers can be given as a comma separated list.

closed 2020-11-21 by hneemann

hneemann, 2020-11-22:
@holstvoogd 
If I think about your question, it would be a good opportunity to add a new toolchain configuration that supports the Arty-A7-35/100 boards.
Have you already worked on this?

holstvoogd, 2020-11-22:
@hneemann 
Ah, cool. I hadn't considered that, it's perfect for this case. 
 
Yes, I've created a config based of the other digilent board & mapped out part of the I/O. Still have a bunch more pins to do though.  I also need to test/debug the clock generation, it seems like the clock on the FPGA runs slower than in simulation.

I'll make a PR when I have the config completed & tested!

commit f54a4639 (2020-11-21): improved error message if the PinControl component is used the wrong way; closes #564