hneemann/Digital #566: Absolute path for data memory

d-teif, 2020-11-26:
I noticed that a path write for memory content (ROM or EEPROM) is absolute from root.  "/users/....." . 
That is not very convenient when you share your file with students. 
Perhaps it will be more efficient to write a relative path from .dig component itself. 
So we can put .hex file file in a same or sub directory than circuit itself.

Thanks 

DT

hneemann, 2020-11-26:
Why should this be a problem. In ROM and EEPROM the contents of the memory are stored in the .dig file.
Simply uncheck the reload checkbox.
Or have I missed something?

d-teif, 2020-11-27:
Its OK with RAM with checkbox
But for EEPROM, the content could be change, and it will be useful to reload the initial content from file.
In this case, you have to load again file from target machine because absolute path is wrong in the design.

Specially when you have lots of designs to verify it will be very boring! If relative path  to hex file is stored in the design, reload will work.

hneemann, 2020-11-27:
But the EEPROM has no "reload" check-box like the ROM has. 
Do you use the preload option from the circuit specific settings? Then I misunderstood you.
Here the problem was already solved. Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) version.
However, the data file must be located in the same folder as the *.dig file.

d-teif, 2020-11-27:
Yes but EEPROM a Reload button.

In design file 
          <string>lastDataFile</string>
          <string>/Users/.../inst.hex</string>  Absolute Name 
        </entry>
 
When you try to reload it on other computer with different system file, it's fail, even if data.hex is in a same folder.

[image]

hneemann, 2020-11-27:
@d-teif 
I must have thought too complicated the whole time! :-)
Please try again the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

commit b292ebdb (2020-11-27): adds the FileLocator to the DataFieldEditor load button, closes #566
closed 2020-11-27 by hneemann