hneemann/Digital #567: How to test for multi-bit outputs?

phpeterson-usf, 2020-11-27:
Thanks for the great tool. Questions about scripted testing:

1. Is it possible to script testing of an output on a sequential circuit like a processor simulation? It's not clear when the output would be tested against the expected value. Is it when the clock stops, or on the first rising edge of the output, or some other trigger?

2. If scripted testing of a processor is possible, what's the correct way to test for multi-bit outputs, like the value of a register? I have an 8-bit register R0 which is an output on the circuit, with an expected value of 48. After reading the Help, I've tried various combinations like:
```
INPUT R0_7 R0_6 R0_5 R0_4 R0_3 R0_2 R0_1 R0_0
0     bits(8,48)
```
For the above, Digital says "R0_7 not found in the circuit"
```
INPUT R0
0     bits(8,48)
```
For the above, Digital says "expected 2 but found 9 items"
```
INPUT R0
0     48
```
For the above, the test always fails. 
I'm sure the R0 output contains 48 when the clock stops, and no rising edge until then.

hneemann, 2020-11-27:
The statement `bits(8,48)` is used if the circuit contains 8 single bit ouitputs which should be 48 in combination.
If the circuit has a single 8 bit output `R0` something like `R0_1` is not available for testing. In this case 

```
INPUT R0
0     48
```
is the the easiest way to test the signal.

If a test line contains a clock identifier `C`, the following happens: First all inputs are set, which are not clock signals. Then the clock signal is set to 1, then back to 0 and after that all outputs are checked. This looks like this:
```
Clk   R0
C     48
```
If no clock identifier `C` is used, nothing special is done with the clock.

phpeterson-usf, 2020-11-27:
Thanks for that information. If the required number of clock cycles is not known, should I just use an arbitrarily large number?

hneemann, 2020-11-27:
The specified number of clock cycles is actually executed. If the number is very large, it will take a long time.

Is there an output that indicates that the result is finished? Then you could clock forward until this notification using the `while` statement. The help text shows an example of how to do this.

phpeterson-usf, 2020-11-27:
Much better. Thanks again for your help.

closed 2020-11-27 by phpeterson-usf

hneemann, 2020-11-29:
@phpeterson-usf 
If you are serious about testing processors, the latest release does not work very well.
This has been bothering me for a long time because my own processor is not tested very well either.
So I implemented some functions that make it easier to test processors.
These functions are included in the latest [pre release](https://infdigital.dhbw-mosbach.de/).

Examples for the new possibilities to write tests especially for processors can be found in the branch "newProcessor" in the file 
[ProcessorTest.dig](https://github.com/hneemann/Digital/blob/newProcessor/src/main/dig/processor/ProcessorTest.dig).

phpeterson-usf, 2020-11-29:
Thanks, is there documentation for how init and program behave? Might be helpful to add fine-grain test cases next semester.

We (usfca.edu) teach a class where the students build up a simplified ARMv7 processor in Digital. Up to now, it has been tested manually by uploading machine code to Digital ROM components and watching the various processor elements run. Our implementation uses a "program number" input to select the ROM that PC reads from.

I'm adding coarse-grain automated testing for this, as we do for C programs. I found that all I had to do was:
1. Add a label to the CLK so I can refer to it from a Digital test component
2. Start the processor with the CLR line low 
3. Add outputs for the "done" signal and R0
and then I can automate test cases using what you showed above. It's a start, though I'm sure per-instruction testing would be helpful as students incrementally develop their processor implementation.

hneemann, 2020-11-29:
There is no documentation yet. First I want to finish the test cases for the example processor to see if everything that is needed is available.

So far, there are three new instructions for initializing the model before starting the test execution:
1. `program` initializes the program memory
2. `init` assigns a value to a signal, where registers can be set too, if they are flagged as measurement values.
2. `memory` writes a value into a RAM to test memory commands.

This allows you to test a processor on a per-instruction basis without having to connect much additional hardware.

Knochi, 2024-02-12:
GOT IT TO work! Just posting here if anyone else searches for similar solution

`repeat(9) C 0 (n+1)`