hneemann/Digital #568: Visual glitches after usage of dropdown menu of component context menu

BenUnderscore, 2020-11-29:
There is a visual glitch that happens when I use a dropdown menu in a context menu. This would be acceptable, but after I close the menu, hoverying over any button (in the ribbon bar at the top) draws the component grid with it at the top left.
[image]
[image]

I'm running Windows 10 10.1.19041 with Digital v0.25 and JRE v8u271, but I have confirmed the bug to exist on earlier versions of Digital as well, as far back as 0.16. Running the noD3D version seems to behave the same.
I can confirm that the register, counter and adder have this problem, but for whatever reason not the ROM.

hneemann, 2020-11-29:
seems similar to #227

hneemann, 2020-11-29:
The problem is that these dialogs are implemented using the unmodified Java Swing components. 
Nothing special is done with them. So I don't think that the problem can be solved that easily.
Does the problem always appear or only from time to time?

BenUnderscore, 2020-11-29:
It appears reproducibly every time, which makes the program basically unusable unless it can be fixed. If it's a Swing issue maybe using a different version of Java might help?

hneemann, 2020-11-29:
That would have been my next advice. JDK 14 and 15 I have tested both successfully. I personally still use JDK 11.

hneemann, 2021-02-05:
See #227

closed 2021-02-05 by hneemann