hneemann/Digital #570: Blank window on dwm

HassanZahin, 2020-11-30:
Digital is working fine on kde plasma. But when I try to open it in dwm, it just shows a blank window.

hneemann, 2020-11-30:
Please read this [thread](https://bbs.archlinux.org/viewtopic.php?id=251721) and let me know if it solves the problem.

hneemann, 2020-11-30:
This is maybe the same issue as #547

HassanZahin, 2020-12-02:
> Please read this [thread](https://bbs.archlinux.org/viewtopic.php?id=251721) and let me know if it solves the problem.

Thanks, it solved it.

> This is maybe the same issue as #547

I don't think so as he is using gnome and gnome is re-parenting. I'm not sure about the wayland part though.

closed 2020-12-02 by hneemann

HassanZahin, 2020-12-02:
@hneemann Just checked in plasma wayland. Seems to work fine

hneemann, 2020-12-02:
> I don't think so as he is using gnome and gnome is re-parenting.

I have already thought the same. But the immediate closing of the menus would fit perfectly.