hneemann/Digital #571: Timing diagrams again

eliask05, 2020-12-04:
This is really a continuation of #359.

I understand that Digital does not have the concept of time and is (I assume) purely event-driven. As noted before, it is not therefore possible to see a "true" timing diagram.

As a quick (?) fix, is it possible to have a "clocked" graph? I.e., a graph that does not update the signals until clocked by an external clock? That way you can change all the inputs you want, without changing the graph and then have the graph update when clocked, either automatically or (usually) manually.

Logisim-evolution does this with their chronogram and really this is the only thing preventing me from using Digital in my labs. Students freak out when they see non-symmetrical clocks or signals jump when they did not expect them to jump.

Thanks!

hneemann, 2020-12-05:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) build.
I have added a new triggered scope. Up to now there is no special shape for it. 
But please also try the old scope because there is a bug in v0.25 which sometimes causes a unexpected result. (See  #551)

eliask05, 2020-12-05:
I did not have a chance to test it extensively but from what I have seen so far, BEAUTIFUL! It does exactly what I had in mind.

THANK YOU for this quick enhancement.

commit ac0d9db8 (2020-12-05): adds a triggered scope, closes #571
closed 2020-12-05 by hneemann