hneemann/Digital #572: Feature request: autogenerated timing diagrams from truth tables

eliask05, 2020-12-08:
Digital automatically generates truth tables for combinational circuits. Is it possible to add a feature that presents the same information in the form of a timing diagram upon request?

hneemann, 2020-12-08:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

hneemann, 2020-12-15:
At the moment, I tend not to merge this feature, as this diagram can be very confusing when the table describes a state machine.
The complete timing diagram may contain the states in a different order than they occur in the state machine, and may also contain states that do not occur in the state machine at all.

eliask05, 2020-12-15:
Understood and agreed. I was talking about combinational circuits, only.

closed 2020-12-16 by hneemann