hneemann/Digital #579: Subcircuit Symbol Too Small

SteveMillman, 2020-12-18:
I've got a simple sub circuit with two inputs and two outputs. I instantiate it in its parent module and the box is so small that the input and output names overlap despite being only three characters. Is there an easy way to get Digital to make the box bigger? The bottom input is INC and the bottom output is CRY. You can see that the Cs are on top of each other.

[image]

It really isn't an option to have students editing the SVG files, so I'm hoping I've missed something important. In the sub circuit, the inputs and outputs are spaced apart vertically, so I don't think it's because they are too close together. Zooming in  doesn't help either - the letters grow with the box. Any suggestions would be helpful.

Thank you,

Steve

version: v0.25

hneemann, 2020-12-18:
Open the sub circuit and adjust the width in the circuit specific settings (Edit->Circuit specific settings)

SteveMillman, 2020-12-18:
Perfect! Thank you!

closed 2020-12-18 by hneemann