hneemann/Digital #580: Memory modules get stale between runs

gbizzotto, 2020-12-19:
When running/debugging a circuit, one can open RAM chips to see their values.
After a simualtion has ended and was restarted, the still-opened windows loose their link the the freshly booted RAM modules. The user has to close and reopen them to get the values for the new run.
Expected: No need to reopen the RAM modules, they link to the newly instantiated circuit.
For clarity, they could be closed if the RAM module is deleted or cut.

hneemann, 2020-12-19:
This behavior is intentional. It allows to compare the RAM contents of several simulation runs. To let the user know that the window is no longer updated, the color of the content is changed when the simulation is finished and the link to the model is lost.

gbizzotto, 2020-12-19:
Ah sorry about that.
It's useful but still confusing.

closed 2020-12-19 by gbizzotto