hneemann/Digital #581: How is FSM instantiated in design for simulation?

svenn71, 2020-12-19:
I have looked for documentation about how to simulate FSM, but only thing the FSM editor can export is SVG.

hneemann, 2020-12-19:
The simulation of an FSM works somewhat unusual. :-)
First, the FSM is analyzed and transformed into a state transition table. (Create -> State Transition Table) 
This table can then be realized by a circuit. (Create -> Circuit)
This is a rather theoretical approach, but this intermediate step should help the students to understand how this process works.

svenn71, 2020-12-20:
Thank you, now I understand.
I hit something I would call a strange behaviour:
When I use underscores in states, tx_send, rx_receive, init_data etc,
then whatever comes behind the underscore is subscripted in the FSM. I quess this is LaTex notation.
If I escape those underscores with \ like in LaTex, the labels are aligned, but I get an error when trying to convert to transition table.

hneemann, 2020-12-20:
Everyone wants to use a backslash in variable names! (See #582) :smile:
Why don't people use CamelCase like I do? :smile:

closed 2020-12-20 by hneemann