hneemann/Digital #582: Export Name with Underscore to Verilog

SteveMillman, 2020-12-20:
I understand that in order to name an input or output with an underscore, you need to escape the underscore like this: my\_signal and it will display nicely as my_signal. However, when exporting to Verilog, it exports as \my\_signal which, unfortunately, is nonstandard and unnecessary since underscore is not a special character in Verilog. Is there a way to get Digital to format the output to Verilog so the backslash is not included? I've searched the discussions and the manual but have found nothing on this. Any help on this would be appreciated.

Thank you,

Steve

SteveMillman, 2020-12-20:
I should add that leaving the signals escaped as written to Verilog leads to things like this in GTKWave:

This is just ugly... Being unable to use underscores in Digital will be a real pain. I'm hoping there is a solution.

Thank you,

Steve
[image]

closed 2020-12-20 by hneemann

SteveMillman, 2020-12-20:
Very nice - I've tested the pre-release version and verified, as I'm sure you did, that it works. Students will start installing this on January 6. Will a new, stable, release be made before then? Or do you think it ok for them to sue the pre-release version?

Thank you,

Steve

hneemann, 2020-12-20:
All test cases and the Verilog/VHDL integration tests are also carried out for each pre-release and are "green".
The main difference between the pre-release and the major release are the GUI integration tests. 
Since these run for a long time, I don't start them before each pre-release.
However, I just ran the GUI tests and they are "green" as well. :-)
Therefore, technically speaking, the latest pre-release  (v0.25-209) is a major release.

I don't think there will be a real major release at January 6.

SteveMillman, 2020-12-20:
That's great. We'll go with this pre-release. I really like Digital!

Thank you,

Steve

SteveMillman, 2020-12-31:
There's another spot which could still be addressed. If a label is applied to a subcircuit, the \ still appears in the comment in the exported Verilog. So if a subcircuit is instantiated multiple times and you search for the label in the Verilog, the \ has to be dealt with. Could this also be addressed via the same or similar fix?

Thank you,

Steve

commit d03cc1c4 (2020-12-20): minor improvement of verilog identifier renaming, closes #582