hneemann/Digital #587: I can haz Flip?

gbizzotto, 2020-12-21:
There's Edit > Rotate.
Flip would come handy.
Just one flip. No need for vertical AND horizontal. One is a rotation of the other, and we already have rotate.

xotmatrix, 2020-12-22:
This would be nice. It would also be nice if it could flip components that have the option to, such as drivers and plexers.

hneemann, 2020-12-22:
Some components already have a "mirror" option. Only the components that already had another option to achieve the same appearance do not have it, because that would not make sense. 
And for compatibility reasons I can't remove the older "Flip Selector Position" option.

gbizzotto, 2020-12-22:
I was thinking about the wires. Sometimes I have a bunch of, say, horizontal wires that all start on the left at the same X, and stop on the right at incrementing Xs, so I can redirect all of them up (or down) in a parallel fashion. That takes time to draw, even with copy/paste.
Sometimes I want to flip them all, so, if they're set up so i can redirect them up, i'd like to now redirect them down. Without flip, the only way I can do that afaik is by deleting  and redrawing everything.

The picture below illustrates this.

[image]

While I was drawing this example, I realized I can't even copy, rotate and paste those wires to redirect them up. I already need to flip them.

[image]

hneemann, 2021-02-07:
There are components like Text, Rectangle and others where a flip makes no sense. 
If such a component is in the selection, there is no elegant way to deal with this.

closed 2021-02-07 by hneemann