hneemann/Digital #592: Flip-Flop Not Respecting Inverted Inputs

SteveMillman, 2020-12-28:
I'm using release v0.25-209-gd03cc1c and I'm getting strange behavior. I'm using the D flip-flop with asynchronous set/reset but wanted active low set/reset. So, I inverted those inputs and get this circuit:

[image]

And this looks exactly right. I then simulate in Digital and all works as expected. If set is low, the output goes high, etc. And then I export to Verilog and it doesn't work. It appears that the inversions are ignored. Here's the Verilog I get for this circuit (just for the flop primitive):

//************************************************************
module DIG_D_FF_AS_1bit
#(
    parameter Default = 0
)
(
   input Set,
   input D,
   input C,
   input Clr,
   output Q,
   output \~Q
);
    reg state;

    assign Q = state;
    assign \~Q  = ~state;

    always @ (posedge C or posedge Clr or posedge Set)
    begin
        if (Set)
            state <= 1'b1;
        else if (Clr)
            state <= 'h0;
        else
            state <= D;
    end

    initial begin
        state = Default;
    end
endmodule

//************************************************************

It should be:

    always @ (posedge C or negedge Clr or negedge Set)
    begin
        if (!Set)
            state <= 1'b1;
        else if (!Clr)
            state <= 1'h0;
        else
            state <= D;
    end

(Note that 1 is missing in the original for the Clr clause.) I checked the VHDL output and it appears to have the same behavior as the Verilog... Can you please confirm if exporting is supported with inverted inputs on the flops? It would seem best if the Digital simulation and Verilog behaved the same way.

Thank you,

Steve

SteveMillman, 2020-12-28:
For completeness, I verified that if I don't invert the set/reset inputs, then I get identical Verilog to that shown in the prior post. However, Digital simulates it correctly as active high set/reset. (And note that I've been saying set/reset but the pins are actually set/clr.)

hneemann, 2020-12-28:
Digital replaces a component with inverted inputs by adding inverters to the circuit and using the unmodified component.
So your circuit becomes 

[image]

And there was a bug in the code that does this substitution, which only has an effect if the component does not simplify to a boolean expression.
Thanks for pointing this out.

commit 426eb6e2 (2020-12-28): fixes a bug in the hdl input inverter handling, closes #592
closed 2020-12-28 by hneemann