hneemann/Digital #593: iverilog path for external component

SteveMillman, 2020-12-28:
I'd like to be able to have the initial contents of a RAM exported via Verilog. But that doesn't seem to work. (I can get a ROM to do it, but I'd like to be able to do a RAM... This would be very useful since we could start from a known spot rather than having to reload the RAM every time. The goal is to export and simulate with iverilog.)

So, I thought I'd try doing an external component. I think I have it set up correctly, but I'm getting an error message that iverilog can't be found:

[image]

I believe I've set up the path correctly in settings. It matches what I see if I execute "which iverilog" in my terminal:

[image]

I'm on a Mac running Catalina using the latest pre-release.

Thank you,

Steve

hneemann, 2020-12-28:
Digital expects that there is an executable file called "iverilog" and starts it. 
This works very well under Linux and Windows. What happens under macOS I can't say, because I never worked with macOS.
But using the iverilog integration is not such a good idea. It works very well, but it is slow, because for the communication between Digital and iverilog OS pipes are used. 

Maybe [this](https://github.com/hneemann/Digital/issues/348#issuecomment-535895594) circuit is a solution.

SteveMillman, 2020-12-28:
I did see that example of having a RAM/ROM combination. However, that won't work if we want to be able to modify the locations in the ROM. I'm curious - why not just add this line to ROM/RAM modules when you export them if they are being loaded from files:

  initial
  begin
     $readmemh("ram_vals.hex",memory);
  end

where "ram_vals.hex" is the file we've told Digital to initialize the RAM or ROM.

The only issue with that is it appears you require the first line to be "v2.0 raw" which has to be commented in/out depending if you are in Digital or in Verilog. This would work nicely for both ROM and RAM instances... That way, changes to the "ram_vals.hex" file when running in Verilog would be automatically reflected if you went back to Digital... (If the v2.0 raw issue is resolved...)

I noticed that for the ROM, you actually have the initial block set the value for every line of the ROM. Using $readmemh would be much cleaner...

I'm sure there must be a number of reasons you didn't choose this method, so I'll probably have to have my students add the above code to their resulting Verilog even though it will mean hacking the Verilog netlist created by Digital.

Thank you,

Steve

hneemann, 2020-12-30:
When the VHDL/Verilog export was planned, the priority of readability was quite low. The goal was to generate code that could be synthesized. Why would a human ever look at this code? It would not be possible to modify it in any case, because then it could not be generated again, because the changes would be overwritten.
It was more important to make the handling of the generated code as simple as possible. And here it seemed to make sense to export all the code into a single file. In the meantime the readability has improved, but I still think that readability is less important.

To me, the hdl code only serves as a vehicle to get a circuit onto an FPGA. Not as something that people look at.

It was also never the plan to utilize all the possibilities of an FPGA. Digital is not an HDL design tool, and never will be. 
The goal is only to be able to export as much Digital circuits as possible to an HDL.

SteveMillman, 2020-12-31:
Understood, and no worries. Thank you!

closed 2021-01-05 by hneemann