hneemann/Digital #594: Reset to Default zoom

edson-acordi, 2020-12-28:
It would be good to have the zoom number in percentage or an option to restore the default zoom (100%).

hneemann, 2020-12-28:
What is the use case?

edson-acordi, 2020-12-28:
> What is the use case?

Hi hneemann, how are you?

Is only a suggestion to an enhancement.
For example, when I prepare classes, I don't know which is the zoom that I have used last time.
This feature is good to get always the same size picture!
Thank you.

hneemann, 2020-12-28:
> Hi hneemann, how are you?

I am doing well, thank you! Hopefully you too! :smile: 

> This feature is good to get always the same size picture!

When creating images (png or svg), scaling is ignored and a gate always has the same size already.

edson-acordi, 2020-12-28:
> > Hi hneemann, how are you?
> 
> I am doing well, thank you! Hopefully you too! 😄
> 
> > This feature is good to get always the same size picture!
> 
> When creating images (png or svg), scaling is ignored and a gate always has the same size already.

How nice! I am ok too! 👍 😄. Yes, when exporting, I know that scaling is ignored. This new feature would be only to rapidly capture screenshoots, without create a file (printscreen). But is ok! The exporting to file is much more useful, mainly, when we are writing a paper.
Again, congrats for the wonderful software that you have created! I can't get enough of saying that!

hneemann, 2020-12-29:
If you are using LaTeX to create your documents, you should try the svg-Inkscape workflow.
This will create a pdf with the graphic and a TeX document with all the text strings. This way the font in the graphic matches the font of the surrounding text. This looks very nice! :+1:

commit b7a76162 (2020-12-29): adds a "set to scale" function, closes #594
closed 2020-12-29 by hneemann