hneemann/Digital #596: Hierarchical Memory Not Initialized

SteveMillman, 2020-12-31:
If a RAM is located at the top level of a design, the preload mechanism works beautifully and simulation works as expected. However, if the RAM is in a sub circuit, then you still need to go into the Circuit specific settings for the top level and set up the preload even though that has already been set up in the subcircuit's Circuit specific settings. Is this expected behavior? What does this imply if there are multiple RAMs/ROMs?

Thank you,

Steve

hneemann, 2020-12-31:
Only memory marked as program memory can be written to in the circuit-specific settings, and there must be only one program memory in a circuit for this to work. If there are multiple program memories, an error message will appear. If there are several program memories, and they all have a different label, the memories are sorted by the label and the file is distributed to the memories, so that you can build a 16-bit program memory from two 8-bit ROMs, for example.

If a subcircuit contains a ROM component, but this subcircuit is used several times, so that there are then several instances of this ROM in the circuit, a different ROM content can be defined for each instance in the circuit-specific settings. This is necessary if several 28cxxx DIL-EEPROM are to be used, which all have a different content. Warren Toomey's "Crazy Small CPU" is an example for this: This CPU contains as program memory, as control unit and as ALU each a 28c64 EEPROM and all three devices have a different content, but it is always the same subcircuit.

All these modifications of the circuit are carried out when the circuit has already been completely transferred to a model that contains only built-in components that are wired up accordingly. And in this state only the current circuit is available. One could of course try to implement a hierarchy of preloading, but I think that this is hardly worth the effort, since there are barely any use cases.

And the idea of a predefined RAM content is a typically HDL specific feature. In classic digital technology there is no such thing. 
In fact, it is taught that RAM is volatile memory that has no defined content when it is powered up. And this is how it should be in the simulator. There is only the one exception for the program memory, which is also needed e.g. to get a program from an assembler IDE into the CPU, so that one can program and debug the CPU more easily.

SteveMillman, 2020-12-31:
I think I understand, but want to be sure. Based on what you've said:

1. There can be only one RAM in the design, and it can be anywhere in the hierarchy but the path to the preloaded values must be specified at the top-level design.

2. There can be a ROM in the design in addition to the RAM and the ROM can still have its preloaded values. (As opposed to having to hand-edit it before simulating.)

If the above conclusions are true, then Digital will meet all of our needs as is and this ticket should be categorized as question and closed.

Thank you,

Steve

hneemann, 2020-12-31:
> There can be only one RAM in the design, and it can be anywhere in the hierarchy but the path to the preloaded values must be specified at the top-level design.

This is not quite correct: There may be several RAMs, but only one of them can be flagged as program memory, and can be initialized with a program.

> There can be a ROM in the design in addition to the RAM and the ROM can still have its preloaded values. (As opposed to having to hand-edit it before simulating.)

There can be any number of ROMs in a circuit, and each can have its own content. Usually this content is stored in the ROM itself and is editable as an attribute of the ROM.
Only if the ROM is located in a subcircuit, which is included several times, this does not work, if the content should not always be the same. In this case the content can be assigned individually via the circuit-specific settings.

SteveMillman, 2020-12-31:
Perfect. Thank you for the clarifications!

closed 2020-12-31 by SteveMillman