hneemann/Digital #599: No tooltips shown for shallow copy in CircuitComponent

maltaisn, 2021-01-03:
There are no tooltips shown for a generic circuit during simulation. It seems that the `CircuitComponent.getTooltipText` only looks at `getCircuit()` and not `shallowCopy`. Making this change fixes the issue.

hneemann, 2021-01-03:
Are you modifing the description of an input or output?

maltaisn, 2021-01-03:
The tooltips for inputs and outputs are shown correctly yes, but these are the tooltips for the generic circuit, not the concrete one. If for example you have an input with the description "foo", but its generic code changes it to something else:
```
this.Description="bar";
```
During simulation the tooltip will show "foo" and not "bar". This example is probably not meaningful though.
The real issue here is that wire tooltips aren't shown during the simulation, and those are pretty useful.

commit 132d5c46 (2021-01-03): fixes an issue with tooltips in generic circuits, closes #599
closed 2021-01-03 by hneemann