hneemann/Digital #600: Is there a way to implement a LCD

edson-acordi, 2021-01-04:
Hi @hneemann, how are you doing?

Is there a way to implement a LCD (with initialization codes)?
I tried to use the terminal, but there is no way to positioning/clear the screen.

Thank you!

hneemann, 2021-01-04:
I'm fine, thank you! :-)

ASCII 0x0c (Form Feed) deletes the terminal, and 0x08 (Backspace) deletes the last character.
If you need more escape sequences, they can easily be added.

An LCD display is difficult to implement, as you would have to mimic an HD44780 or similar.

edson-acordi, 2021-01-05:
That's good! :)

Ok, using ASCII, some command can be done. Thank you.
It seems there is a bug on the field to enter the number of lines if you try a number less than 5.

hneemann, 2021-01-05:
This is not a bug. The minimum number of lines is set to 5. 
Less did not seem to make sense to me. What number do you want to use?

Is deleting the terminal enough for your needs or are more escape sequences required?

edson-acordi, 2021-01-05:
Ok, I thought that was a bug because I can enter a number less than 5, but ok.
Thank you.

closed 2021-01-06 by hneemann