hneemann/Digital #605: Generate wiring lists based on common IC families

mitolegy, 2021-01-10:
Hi, me again. This might be a rather large request idk. But I was recently reminded of a pretty nice feature the simulation software Proteus had. Wich would be super useful when moving your circuit from simulator to reality. 

You see proteus could generate wiring lists based on common IC layouts. So it would number the pins and name components like the AND, OR, and NOT gates according to their real life 4000 series or 7400 series counterparts.

So for example the first gate would have the name AND_1 and it's pins would be 1,2,3. Then the next one would be named the same but have PIN numbers 4,5,6. Next one 9,10,11 due to the ground and VCC pins. And so on. Since there's only 4 AND gates in a single 7408 package. The fifth one would be called AND_2 since it's a new separate package. and have PIN numbers 1,2,3 again. 

Again a realize this is probably a large request. But it was super useful when actually building the PCB layouts and such. As I didn't have to essentially rebuild the circuit from scratch. Just follow the diagram. connect pin 6 on AND_14 to pin 1 on OR_11 and so on. 

Even if it is just a basic list of pure logic components. No support components like proteus had. It's still greatly improve the process of realization.

hneemann, 2021-01-10:
You are right. It's complicated because you have to carefully choose which gates to place together in a chip to make wiring easier. In addition, most components cannot be easily mapped to standard ICs.

But building a more complex circuit with standard ICs is tedious, error-prone, expensive, and requires a large PCB. Wouldn't it be better to use a CPLD or FPGA?

And if you really want to use standard ICs, wouldn't it be easier to start with them from the beginning instead of using generic gates at first? Many of the 74xx ICs are already available in Digital.

mitolegy, 2021-01-10:
> You are right. It's complicated because you have to carefully choose which gates to place together in a chip to make wiring easier. In addition, most components cannot be easily mapped to standard [...]

Ah but you see. FPGAs are no fun I like it when my circuits look like something straight out of the 70s. And also. Building with the 74xx components already in the simulator get confusing and really messy really fast. Even just a simple 1-bit Register becomes an ungodly mess of wires within seconds. Hard to read and understand and even harder to debug. 

But it doesn't have to be very complex ether. First off not every component needs to have a real life equivalent. Maybe just the basic logic components. The flip-flops too if you feel so inclined. Things wich have an obvious equivalent IRL. And secondly. The way proteus does it is that it just numbers it as you place them down. Since you build a circuit piece by piece. Rather then jumping around randomly. The components wich need to be near each other tend to cluster together kinda naturally.

Multi bit components do pose a bit of a challenge. But you could make it available only to single bit components. Or split each bit into its own separate physical component.

hneemann, 2021-01-10:
> Ah but you see. FPGAs are no fun

I'm probably too much of a software guy. :smile: 
I found it so impressive when I saw my own processor running on real silicon at 20MHz along with a self made VGA card! And it was so easy to try and change things! 
But a sea of DILs certainly has its charm, too. Some time ago I read about someone who made a processor out of thousands of TO92 transistors. To each his own!

I will think about this idea.

ddotcole, 2021-01-11:
>Building with the 74xx components already in the simulator get confusing and really messy really fast. 

I think this is the situation where the TUNNEL component works very well.  Its very clean and easy to layout and move around the DIL packages with their respective TUNNELs to reorganize them.

Otherwise, I was actually wondering myself about converting schematics over to PCBs.  In my case I have built my schematic using the DIL 74xx chips using TUNNELs to substitute for wiring.  It would be oh so convenient if one could export this to a PCB drawing package so you can build a board and send it to some of these quick turnaround PCB manufactures.

mitolegy, 2021-01-11:
> I think this is the situation where the TUNNEL component works very well.  Its very clean and easy to layout and move around the DIL packages with their respective TUNNELs to reorganize them. [...]

The tunnel component probably does make things a bit easier to understand . I will try that. But it's still not as easy to read as native circuit diagrams, and when you're building say an ALU, building and debugging is still a pain. Not to mention slow. And it doesn't eliminate the problem if essentially having to entirely rebuild your circuit from scratch using DILs