hneemann/Digital #612: Updating Chinese Translation

hneemann, 2021-01-14:
@itviewer 
in the near future I would like to publish a new release. Therefore the request for updating the translation.

There is a complicated change, because the test cases have been extended significantly. Unfortunately this is a long text and the changes are not obvious.

At the beginning a heading with `<h3>General</h3>` was added and at the end an explanation of the new `declare` statement was added. After that you will find the heading `<h3>Processors</h3>` with a new short chapter.

It would be nice if you could complete the translation.

[lang_diff_zh.xml.zip](https://github.com/hneemann/Digital/files/5813115/lang_diff_zh.xml.zip)

itviewer, 2021-01-14:
Here is the pull request https://github.com/hneemann/Digital/pull/614, so you can merge directly.
Since the content of msg_testVectorHelp is too much, I have not translated it, and kept it as it is.
I think most of the content except msg_testVectorHelp has been translated.

hneemann, 2021-01-14:
It is very hard for me to keep the translations consistent when I merge a pull request.
It is much easier if the above XML file is edited and then reimported by me. 
Just remove the key msg_testVectorHelp from the XML file if you don't want to translate it. That' s perfectly fine.

itviewer, 2021-01-15:
[lang_diff_zh.zip](https://github.com/hneemann/Digital/files/5820169/lang_diff_zh.zip)

hneemann, 2021-01-15:
Thank you for the double effort. :+1:

commit f18f88f5 (2021-01-15): updates the chinese translation, closes #612
closed 2021-01-15 by hneemann