hneemann/Digital #613: Updating French translation

hneemann, 2021-01-14:
@maltaisn 
in the near future I would like to publish a new release. Therefore the request for updating the translation.

There is a complicated change, because the test cases have been extended significantly. Unfortunately this is a long text and the changes are not obvious.

At the beginning a heading with `<h3>General</h3>` was added and at the end an explanation of the new `declare` statement was added. After that you will find the heading `<h3>Processors</h3>` with a new short chapter.

It would be nice if you could complete the translation.

[lang_diff_fr.xml.zip](https://github.com/hneemann/Digital/files/5813124/lang_diff_fr.xml.zip)

It is easiest if you send the translated file back to me by email or as a reply to this ticket.

maltaisn, 2021-01-14:
Sure, I'll have some time later today.

maltaisn, 2021-01-14:
[lang_diff_fr.zip](https://github.com/hneemann/Digital/files/5816665/lang_diff_fr.zip)

commit 3827aaa2 (2021-01-15): updates the french translation, closes #613
closed 2021-01-15 by hneemann