hneemann/Digital #617: Generic circuit recursion limit

YulienPohl, 2021-01-19:
I found out that if you have a recursion greater than 29 in a generic circuit Digital throws an error.

> Error creating the circuit
The circuit Delay_Final_Inc.dig imports itself!
Occurred in file Delay_Node_Inc.dig.

I have attached an example which implements a [Delay Register.zip](https://github.com/hneemann/Digital/files/5837700/Delay.Register.zip)

It would be great if we can get a deeper recursion.
By the way setCircuit() for build in circuits would also be good.

hneemann, 2021-01-19:
The  number 30 was not chosen for any particular reason. Only some value is required to detect a self-importing circuit. Now that there is recursion the value is actually somewhat small.

Since such high values actually only result from replacing a loop - divide and conquer would blow up the memory at this depth - you might want to look at the new possibilities that have become available. 
Take a look at this [example](https://github.com/hneemann/Digital/blob/master/src/main/dig/generic/modify/Conway/GenericConway.dig): 
This new way of implementing generic circuits is much easier to use and also allows the use of built-in components.
However, the latest pre-release is required for this.

hneemann, 2021-01-19:
This is how your example would look like: [DelaySimple.dig.zip](https://github.com/hneemann/Digital/files/5838562/DelaySimple.dig.zip)

YulienPohl, 2021-01-19:
Thanks for the clarification.
I hope the new generic circuit approach gets well documented.

hneemann, 2021-01-21:
That is my hope too! :smile:

closed 2021-01-21 by hneemann