hneemann/Digital #620: Digital cannot find iverilog

jay20162016, 2021-01-25:
```
all tests are skipped: iverilog (https://github.com/steveicarus/iverilog) is not installed! Add iverilog binary to the system path or set system property 'iverilog' to iverilog binary
```
iverilog is in path.

jay20162016, 2021-01-25:
Ok, I installed iverilog via homebrew, so it was a relative symlink (`../Cellar/blah/iverilog`). Making it an absolute link (`/usr/local/bin/Cellar/blah/iverilog`) worked.

hneemann, 2021-01-25:
This error message only appears when Digital is built from the sources. As long as you are not working on the Verilog templates or the Verilog code generator, you can ignore this error message. All tests are always "green" in master.

closed 2021-01-25 by hneemann