hneemann/Digital #629: Issue with panning the view

DanHarty, 2021-01-29:
By panning, I mean dragging the screen around with right-click-and-drag. 

The issue is that once you begin a selection by left-click-and-drag, right-click-and-drag no longer pans the view. Instead, it begins a new selection. Am I missing something here? Is this a bug, or a confusing design choice? Is it possible to pan the view when making a selection? I might never be able to get used to this. I keep accidentally starting new selections when I am trying to pan the view.

One way to fix this issue would be to make it so that right-click-and-drag only ever does one thing: pan the view. Right-click-and-drag should never start a new selection- that is done by left-click-and-drag. Whereas now, right-click-and-drag sometimes pans the view and sometimes starts a new selection. 

Another option would be to make panning the view also bound to middle-mouse-and-drag. I think this would be good to implement regardless, as it would make the control scheme more consistent, with one mouse button per action: left is select, right is option, and middle is pan. Right-click-and-drag should either do nothing or be kept as an alternative to middle mouse.

PS if someone could point me to where in the code the panning/selecting behavior is defined I'd appreciate it.

hneemann, 2021-01-30:
I could not reproduce exactly what you described, but I too encountered some strange behavior that I never noticed before.

Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) and let me know that feels.

hneemann, 2021-01-30:
BTW: The code that handles the mouse action is located [here](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/gui/components/CircuitComponent.java).

closed 2021-02-03 by hneemann