hneemann/Digital #633: Issue with VHDLSimulatorTest

lmcapacho, 2021-02-01:
When I run `mvn install` to build digital in Ubuntu 20.04 ( Apache Maven 3.6.3, OpenJDK 11.0.9.1, and GHDL 0.37), I get these errors:

Test set: de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest

Tests run: 7, Failures: 0, Errors: 3, Skipped: 0, Time elapsed: 29.933 s <<< FAILURE! - in de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest
testGHDLInSimulator(de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest)  Time elapsed: 0.082 s  <<< ERROR!
java.lang.Exception: errors testing files: 4
	at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.testGHDLInSimulator(VHDLSimulatorTest.java:125)

testInSimulator(de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest)  Time elapsed: 19.136 s  <<< ERROR!
java.lang.Exception: errors testing files: 1
	at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.testInSimulator(VHDLSimulatorTest.java:45)

testProcessorInSimulator(de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest)  Time elapsed: 0.747 s  <<< ERROR!
java.io.IOException: 
应用退出状态非零 1：../../src/openieee/v93/numeric_std-body.vhdl:192:13:@0ms:(assertion warning): NUMERIC_STD.TO_01: non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:88:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: non logical value detected, returning 0
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
../../src/openieee/v93/numeric_std-body.vhdl:901:9:@0ms:(assertion warning): NUMERIC_STD."*": non logical value detected
/usr/bin/ghdl-mcode:error: bound check failure at ../../src/openieee/v93/numeric_std-body.vhdl:1227
in process .hdlexample_dig_tb(behav).main_0@main(behavioral).gate0@processorhdl(behavioral).gate11@alu(behavioral).gate4@dig_neg(behavioral).P0
  from: ieee.numeric_std.has_0x at numeric_std-body.vhdl:1227
/usr/bin/ghdl-mcode:error: simulation failed

	at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.startProcess(VHDLSimulatorTest.java:175)
	at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.runGHDL(VHDLSimulatorTest.java:158)
	at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.checkVHDLExport(VHDLSimulatorTest.java:142)
	at de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest.testProcessorInSimulator(VHDLSimulatorTest.java:88)

hneemann, 2021-02-01:
Is it possible that the test cases have been changed?
On my machine it runs with ghdl 0.37 without problems.
Or are you possibly using a nightly build? I have only tested with the 0.37 release.

lmcapacho, 2021-02-01:
You are right. I tested with GHDL 0.37+dfsg-1ubuntu1 from Ubuntu packages, and I got those errors. I compiled GHDL 0.37 release from the source code, and the build process was successful. 

Test set: de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest
Tests run: 7, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 34.055 s - in de.neemann.digital.hdl.vhdl2.VHDLSimulatorTest

hneemann, 2021-02-01:
In this case, I would leave everything as it is and wait to see what happens when GHDL 0.38 is released.

closed 2021-02-01 by lmcapacho

lmcapacho, 2022-04-29:
I updated the last release of GHDL 2.0.0 and executed `mvn install` The build was a success.