hneemann/Digital #642: Problem when load a .hex file on the version v0.26

edson-acordi, 2021-02-09:
The .hex files saved from an EEPROM in the version v0.25 does not open correctly in the version v0.26.

hneemann, 2021-02-09:
Could you please attach an exmple of a file, which can be loaded with 0.25 and causes problems width 0.26?
I have played arround with some files and everything seems to work like expected.

edson-acordi, 2021-02-09:
Hi @hneemann, sorry about that, I have realized that the problem is with my .hex file.
The problem is caused because some bytes are incomplete (with a nibble - first ten lines of code, but it works in v0.25).

v2.0 raw
0
a
0
a
0
a
0
a
0
a
5*2
10
1a
10
1a
10
1a
10
1a
10
1a
5*12
20
2a
20
2a
20
2a
20
2a
20
2a
5*22
30
3a
30
3a
30
3a
30
3a
30
3a
5*32

hneemann, 2021-02-09:
The problem is the blank in front of the '*12'.

closed 2021-02-09 by hneemann

edson-acordi, 2021-02-09:
Hi @hneemann, thank you for your response!
Ok, I put a blank only for posting in Github. My .hex file does not have these blanks.
I put it here because the GitHub format the writing 05*12 (without blank), for 0512. I did not understand why.
Thus, the blank seems to be a new problem.

hneemann, 2021-02-10:
Please zip a non working hex file and attach the zip file to this issue.
Without being able to reproduce the bug I can't fix it.

edson-acordi, 2021-02-10:
See the attached file.
Thank you.

[STW_LDW_test_Micro2.zip](https://github.com/hneemann/Digital/files/5958955/STW_LDW_test_Micro2.zip)

hneemann, 2021-02-10:
There must be some kind of misunderstanding or misinterpretation. Either on my side or on your side. The file also opens with v0.26 without problems.

Can you describe what happens on your machine and what you would expect instead?

edson-acordi, 2021-02-26:
Hello @hneemann, sorry about the late reply.
I tried to reproduce the error, but it does not happen anymore.
I think that was a problem with my .hex file. Sorry about that.

commit 7032942a (2021-02-09): fixes an issue in the logisim hex reader; closes #642