hneemann/Digital #646: Component request:  74xx299 Universal 8-bit shift register

monsonite, 2021-02-13:
I would be most grateful if you could add the 74xx299 universal 8-bit shift register to your TTL library, when you get an opportunity.

With the 74xx299 it will be possible to create SPI peripherals without having to resort to a combination of 74166, 74595 and 74573.

Thanks in advance,

Ken

I can be contacted at ken.boak@gmail.com

hneemann, 2021-02-13:
Have you already created this chip yourself? 
In that case you can attach the circuit to this ticket and I will add it to the library.

closed 2021-02-13 by hneemann

xotmatrix, 2021-02-14:
I recreated the 74299 universal shift register based on datasheet schematics. There is one weird issue though. The IC has bidirectional pins for parallel input and register output. The datasheets I've looked at say when parallel loading is active, the state of output enable does not matter. I'm not sure how that could work, ie. a pin being an input and an output at the same moment. This can cause a short circuit in the simulation. The chip may have protections but I don't understand how that would work and if it can be simulated in the same way. Short circuits could be prevented with some extra driver logic to disable the output when parallel input is selected but is that the correct solution? I do not have real hardware to test the actual behavior of the chip.

[74299.zip](https://github.com/hneemann/Digital/files/5978595/74299.zip)

monsonite, 2021-02-15:
Hi Helmut and Xotmatrix,

This afternoon I completed the implementation of the 74xx299 shift register
- following the TI Datasheet from 1974 which shows the internal structure.

https://www.ti.com/lit/ds/sdls156/sdls156.pdf  (below)

It can be seen on the left side that the S0 and S1 inputs are inverted and
NORed together in a dual input gate, which is then used to overide the /G1
and /G2 signals using a 3-input AND gate, which forces the outputs into a
tristate (high Z) condition when S0 and S1 are both high for the parallel
load condition.

I have attached my 74299 circuit file.

[image: image.png]

monsonite, 2021-02-15:
Xotmatrix,

I checked out your 74xx299 schematic and tested the S0 S1 logic in a
simulation. Everything is good.

You will find that when S0 and S1 are both high, the states of /G1 and /G2
are ignored and the outputs are put into high-impedence state.

regards

Ken

xotmatrix, 2021-02-15:
Oh, OK, that's weird. It does appear to work correctly after all. Not sure what I was doing wrong with my testing before. I probably had only one selector enabled when the short happened. Thanks for letting me know.

hneemann, 2021-02-15:
I also noticed that the circuit shown in the datasheet i used does not do what is described in the text.
In my case the control signal for the drivers was inverted.

commit cb281353 (2021-02-13): adds 74299, closes #646