hneemann/Digital #648: Request: Less expensive way to display decimal numbers on LED displays

rj45, 2021-02-14:
Hi!

So, some feedback I got on my youtube series is to avoid hexadecimal and use decimal as much as possible. I am currently building a blinkenlight frontpanel for the CPU I am building. And I want to display numbers with 7-segment LEDs instead of output pins for maximum blinkenlights.

So, in order to display in decimal on 7 segment displays, I need to convert a binary 32-bit number into a 10 digit BCD number. Here is the circuit to do that:

[image]

Each blue box contains this:

[image]

Here's the stats on that:

[image]

Now, this is just a front panel, I don't plan to include this as part of the CPU I export to verilog. It's only for debugging. So that's quite a lot of components just to convert a number into a more human friendly one. And I will need multiple of these to display all the numbers I want to display.

BTW, yes you can do this iteratively, but then you have to clock it 32x faster than the CPU itself, and deal with either multiple clock domains or figuring out how to single step a CPU when the debug displays take 32 cycles to update. But for reference this is what that would look like:

[image]

I would _love_ if there was a decimal to bcd converter component that was super fast and didn't require 153 adders and comparators. Or alternatively, a set of LED displays that would work really well for displaying debug info inside a processor in a configurable base (decimal would be valid) and have that nice LED blinkenlight look. Or whatever better idea you come up with.

Thanks! And just an aside: playing around in Digital is a ton of fun! Thanks so much for working on this, it's so very appreciated!

xotmatrix, 2021-02-14:
You could build a 32-bit decimal decoder with 10 ROMs, one ROM for each digit. Or a single ROM in a multiplexed display if they could support 36 bits of addressing.

rj45, 2021-02-14:
Yeah, at least 4 GB per ROM...

I could replace the "> 4 ? +3" circuit with a ROM.... that would probably speed it up a bit? Maybe make a ROM for 3 of them, 2 of them and 1 of them. I still need 30 rows of them though. Maybe I could make a ROM that calculates 3 rows at a time. So there is optimizations I could do to make the circuit have fewer parts / simulate faster.

xotmatrix, 2021-02-14:
Sorry for deleting the unhelpful post you replied to.

hneemann, 2021-02-14:
Why not simply add a probe and select decimal format?

xotmatrix, 2021-02-14:
If a probe (or any other text output) could allow one to set a system font, font size, and font color, this seems like it could be a good presentation solution.

rj45, 2021-02-14:
Yeah, it's mainly about the blinkenlights -- that is, making a front panel that's a bit of eye candy.

hneemann, 2021-02-14:
To be honest, I neither want to add a font selection box, nor a BCD encoder. 
So I would advise to just stick with the current solution with the BCD encoder, or build a custom BCD encoder component if there really is a performance problem. This [repo](https://github.com/hneemann/digitalCustomComponents) would be a good starting point.

rj45, 2021-02-14:
Would you be willing to add more blinkenlight options? Like a probe that looks like an LED display?

I mean, if the answer is no, perhaps some day I would have some time to dig into Java and build it myself. Would you accept PRs for aesthetic features like this by chance?

eliask05, 2021-02-14:
You may want to take a look at a simpler solution that uses flip-flops:
http://www.ingelec.uns.edu.ar/dclac2558/BCD2BIN.PDF
(I couldn't find the original Xilinx application note).

hneemann, 2021-02-14:
> Would you accept PRs for aesthetic features like this by chance?

To be honest, I don't think so! That's why I created the custom component example repository: Everyone can add what they want without having to include everything in the main branch.
This gives users the maximum freedom and at the same time keeps the main branch clean.

On the other hand, a BCD encoder is not that fancy either. Maybe one should add one after all. I'll have to think about it.

rj45, 2021-02-14:
Okay, I totally understand. Feature creep is good to fight against. If you decide against adding the BCD encoder feel free to close this :-)

hneemann, 2021-02-14:
Only 10 components required. :smile:

[image]

Nobody would ever build that in hardware. But in the simulation it works fine! [bcd32.zip](https://github.com/hneemann/Digital/files/5978257/bcd32.zip)

hneemann, 2021-02-14:
Now that it has become clear how easy it is to implement a BCD encoder for simulation, a corresponding component is no longer required.

closed 2021-02-14 by hneemann

rj45, 2021-02-15:
Finally got a chance to look at it, very simple, and exactly what I was looking for. Thanks again!

xotmatrix, 2021-02-15:
Today I learned there is a divide component.

wbecker, 2024-12-31:
There is a 74xxx component to do this that would be nice to have - the 74185. 

Datasheet - https://www.alldatasheet.com/datasheet-pdf/view/102201/NSC/74185.html

Would be nice to have this in the library!