hneemann/Digital #649: Oscillation error in counter

YulienPohl, 2021-02-14:
pre-release version v0.26-64-g59a1577

When the counter value is max the overflow flag oscillates if you have a circuit like this one.
Affected are both counters.

[image]

hneemann, 2021-02-14:
This behavior is exactly what you would expect. This is what such a counter looks like:

[image]

YulienPohl, 2021-02-14:
Okay, this means the only way to stop the counter before overflowing is to use a comparator with 2^n -1.

hneemann, 2021-02-14:
You could also use the ovf output to trigger a FF that stops the counter. This depends on your requirements.

closed 2021-02-15 by hneemann