hneemann/Digital #656: Circuit for checking if signal is undefined

YulienPohl, 2021-02-20:
Currently, there is no way to check if a signal is undefined.
The only way is to use the vhdl function is_x() which returns true
if the value is one of ('U', 'X', 'Z', 'W', '-').

Additionally, it would be useful if constants accept 'Z'.

hneemann, 2021-02-20:
> Currently, there is no way to check if a signal is undefined.

What is the use case?

> Additionally, it would be useful if constants accept 'Z'.

In the "Wires" menu there is a component "not connected" that outputs High-Z.

YulienPohl, 2021-02-20:
I am currently creating complex generic circuits of which the output must be predictable.
When the signal in Digital is 'Z' the output is random.
To counteract this, you check if the signal is undefined if true the output is set to 0, else the output is passed through.

hneemann, 2021-02-20:
Imagine that the circuit is a real circuit, in front of you on the table. Then an undefined floating input would behave exactly like this. When it is read, more or less random values result. If defined values are to be read, a pull-up or a pull-down can be added.

And if you really want to test if an input is floating, a more complex analog circuit is required, which is not the focus of this simulator.

YulienPohl, 2021-02-20:
True, I have been programming in vhdl for so long where STD_LOGIC exists that i have overlooked the most obvious.

hneemann, 2021-02-21:
You can think of it this way:  A hdl describes the function of a circuit. In order to realize this circuit, the HDL code must be converted into a specific technology. This can be individual transistors (ASIC) or a bitstream for an FPGA or a circuit for the simulator. At this level there is no 'U', 'X', 'W' or '-'. Only 'Z' can be realized under certain circumstances, but is avoided, because CMOS inputs do not cope well with high impedance inputs (see https://github.com/hneemann/Digital/issues/36#issuecomment-315969098).

YulienPohl, 2021-02-21:
I'am aware of all.
Our disagreement is that a simulation **does not** reflect the real world.

hneemann, 2021-02-21:
> Our disagreement is that a simulation does not reflect the real world.

That's not a disagreement, that's certainly true.  But I try to mimic the real world as good as possible. 

And for that the std_logic type is not helpful. If you connect a multimeter to a real circuit, you can't measure 'signal hasn't been set yet' or 'don't care'. And with pure digital logic it is even hard to find out if an input has a high impedance.

closed 2021-02-21 by hneemann